PX4/PX4-Autopilot #12260: EKF2 Heading issue with Px4 1.9.0 on Pixracer with External Vision Pose

marcelinomalmeidan, 2019-06-13:
**Describe the bug**
I am using external vision estimation to feed into EKF2 on a Pixracer running Px4 1.9.0. 
I am obtaining vision poses using an Intel T265, hooked up to an Intel Up Core board. 
The Up Core board communicates with the Pixracer through TELEM2 over serial at a baud rate of 115200.
The Up Core board publishes vision estimation at 40Hz into the topic /mavros/vision_pose/pose.
Mavros version is 0.30.0.
I am setting the px4 parameters using QGroundControl.

**Results**
Px4 takes position estimation, but it completely neglects heading information coming from Mavlink.

**EKF2 parameters**
SYS_MC_EST_GROUP: EKF2
EKF2_AID_MASK: 24 (enabled vision position fusion and vision yaw fusion)
EKF2_HGT_MODE: Vision
SYS_HAS_MAG: Disabled

**Alternative results**
The estimation works once I switch to LPE. Hence, it makes me think that there is some issue with EKF2, or maybe I am not setting all the parameters correctly.
Any help or feedback would be appreciated.

siddhu45, 2019-06-28:
Hi @marcelinomalmeidan, I completely agree with you.

I am trying something similar on Pixhawk 4 with external vision fed to EKF2.
I tried firmware 1.8.2, 1.9.0
I am setting these parameters (I don't use GPS)
EKF2_AID_MASK: 24 (enabled vision position fusion and vision yaw fusion)
EKF2_HGT_MODE: Vision
EKF2_MAG_TYPE: None (enabling or disabling this doesn't seem to make a difference in my case)

What I do is I am giving external vision position and yaw reference for the Pixhawk. Setting these parameters gives me the heading I want from the vision system and not the magnetometer heading. I can confirm this from QGC. My vision system and Pixhawk are on the same platform, so whenever the platform rotates (Pixhawk and vision system rotate together), the heading in the QGC changes with it and slowly converges to the external vision yaw. Because of this drift I am not able to arm it immediately and I have to wait for the heading to converge to COM_ARM_EKF_YAW deviation for vision yaw, to arm.

The error message I get is "PREFLIGHT FAIL: EKF YAW ERROR". I read [here](https://docs.px4.io/en/flying/pre_flight_checks.html#ekf-preflight-checkserrors) and found that this is because the firmware checks for a drift between IMU and external yaw reference.

This didn't happen in LPE where you set the ATT_EXT_HDG_M to Vision and that works, the Pixhawk heading always follows the vision system, no deviation. I guess the implementation is different in EKF2, so what other parameters can we set or change to make this work. Are there any weights we put on to trust the vision system more or something like that. Is this actually an issue? It might be related to #11847 
Thanks for any guidance or help

julianoes, 2019-09-26:
Do you have advice here? @kamilritz

kamilritz, 2019-09-26:
> Hence, it makes me think that there is some issue with EKF2, or maybe I am not setting all the parameters correctly.

There was an issue with computation of the transformation between the vision and EKF reference frame. This could also have caused the "PREFLIGHT FAIL: EKF YAW ERROR". This is know fixed on ECL and Firmware side.

I am currently running the identical hardware setup as @marcelinomalmeidan. But I am using the odom plugin in mavros. This is still in a pending [PR](https://github.com/mavlink/mavros/pull/1300) on mavros. I am currently able to run with several EKF_AID_MASK as stated [here](https://github.com/PX4/ecl/pull/642). Expect to have a proper documentation on this in October.

marcelinomalmeidan, 2019-09-26:
Thanks @kamilritz ! I think that my solution back then was to revert to Px4 1.8, which fixed my issue for some reason. Thanks for pointing out the github issue with your tests, that is resourceful information.

closed 2019-10-01 by julianoes