PX4/PX4-Autopilot #12263: Build Firmware error

deepanshgoyal0718, 2019-06-14:
I am building this Firmware using QT Creator by following the steps from "https://dev.px4.io/v1.8.0/en/setup/building_px4.html." When I try to create the project file I get this error. Please tell me the steps to proceed. Is there any other method to build the Firmware of PX4 from source.

$cmake ../Firmware -G "CodeBlocks - Unix Makefiles"

fatal: Not a valid object name HEAD
-- PX4 config file: /home/ceeri2/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- PX4 version: 
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
error: pathspec 'msg/tools/gencpp' did not match any file(s) known to git.
error: pathspec 'msg/tools/gencpp' did not match any file(s) known to git.
error: pathspec 'msg/tools/gencpp' did not match any file(s) known to git.
error: pathspec 'msg/tools/genmsg' did not match any file(s) known to git.
error: pathspec 'msg/tools/genmsg' did not match any file(s) known to git.
error: pathspec 'msg/tools/genmsg' did not match any file(s) known to git.
error: pathspec 'src/lib/DriverFramework' did not match any file(s) known to git.
error: pathspec 'src/lib/DriverFramework' did not match any file(s) known to git.
error: pathspec 'src/lib/DriverFramework' did not match any file(s) known to git.
CMake Error at CMakeLists.txt:371 (add_subdirectory):
  add_subdirectory given source "src/lib/DriverFramework/framework" which is
  not an existing directory.

error: pathspec 'src/lib/ecl' did not match any file(s) known to git.
error: pathspec 'src/lib/ecl' did not match any file(s) known to git.
error: pathspec 'src/lib/ecl' did not match any file(s) known to git.
error: pathspec 'src/lib/matrix' did not match any file(s) known to git.
error: pathspec 'src/lib/matrix' did not match any file(s) known to git.
error: pathspec 'src/lib/matrix' did not match any file(s) known to git.
CMake Error at src/lib/CMakeLists.txt:46 (add_subdirectory):
  The source directory

    /home/ceeri2/src/Firmware/src/lib/ecl

  does not contain a CMakeLists.txt file.

error: pathspec 'src/drivers/gps/devices' did not match any file(s) known to git.
error: pathspec 'src/drivers/gps/devices' did not match any file(s) known to git.
error: pathspec 'src/drivers/gps/devices' did not match any file(s) known to git.
CMake Error at cmake/px4_add_module.cmake:210 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo_lookup".
Call Stack (most recent call first):
  src/modules/attitude_estimator_q/CMakeLists.txt:34 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:210 (get_target_property):
  get_target_property() called with non-existent target
  "df_driver_framework".
Call Stack (most recent call first):
  src/modules/commander/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:210 (get_target_property):
[... 93 lines cut ...]
error: pathspec 'Tools/sitl_gazebo' did not match any file(s) known to git.
error: pathspec 'Tools/sitl_gazebo' did not match any file(s) known to git.
error: pathspec 'Tools/jMAVSim' did not match any file(s) known to git.
error: pathspec 'Tools/jMAVSim' did not match any file(s) known to git.
error: pathspec 'Tools/jMAVSim' did not match any file(s) known to git.
CMake Error at /usr/share/cmake-3.5/Modules/ExternalProject.cmake:1915 (message):
  No download info given for 'sitl_gazebo' and its source directory:

   /home/ceeri2/src/Firmware/Tools/sitl_gazebo

  is not an existing non-empty directory.  Please specify one of:

   * SOURCE_DIR with an existing non-empty directory
   * URL
   * GIT_REPOSITORY
   * HG_REPOSITORY
   * CVS_REPOSITORY and CVS_MODULE
   * SVN_REVISION
   * DOWNLOAD_COMMAND
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/ExternalProject.cmake:2459 (_ep_add_download_command)
  platforms/posix/cmake/sitl_target.cmake:33 (ExternalProject_Add)
  platforms/posix/CMakeLists.txt:117 (include)

-- Configuring incomplete, errors occurred!

julianoes, 2019-06-14:
Something looks broken in your Firmware directory. Did you clone this using git?

Can you do `git status` and paste the output here? And please put pasted output between ```.

deepanshgoyal0718, 2019-06-14:
```
$ git status
On branch master

Initial commit

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	CMakeLists.txt
	CODE_OF_CONDUCT.md
	CONTRIBUTING.md
	CTestConfig.cmake
	Documentation/
	Firmware.sublime-project
	Firmware/
	Jenkinsfile
	LICENSE
	Makefile
	PULL_REQUEST_TEMPLATE.md
	README.md
	ROMFS/
	Tools/
	appveyor.yml
	boards/
	cmake/
	eclipse.cproject
	eclipse.project
	integrationtests/
	launch/
	msg/
	package.xml
	platforms/
	posix-configs/
	src/
	test/
	test_data/
	validation/

nothing added to commit but untracked files present (use "git add" to track)
```

deepanshgoyal0718, 2019-06-14:
Thanks for your help I figured out the problem. There was a problem during cloning. I cloned again using git and it worked this time

> Something looks broken in your Firmware directory. Did you clone this using git?
> 
> Can you do `git status` and paste the output here? And please put pasted output between ```.

julianoes, 2019-06-14:
Ok thanks for the feedback.

closed 2019-06-14 by julianoes