PX4/PX4-Autopilot #12265: telemetry_status advertise failed

julianoes, 2019-06-14:
Recently in jMAVSim SITL I often see this on startup:
```
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-06-14/07_53_57.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
ERROR [uorb] telemetry_status advertise failed
```

Presumably this has to do with uORB publication/subscription changes. FYI @dagar @mcsauder

dagar, 2019-06-15:
I'm not seeing this locally. What are you running exactly? Is everything completely default?

dagar, 2019-06-15:
If you see this again can you quickly check a few things?

`mavlink status`
`listener telemetry_status`
`uorb top -1`

mcsauder, 2019-06-16:
I'm not seeing it locally either:
[image]

julianoes, 2019-06-17:
Ok, it only seems to happen with `PX4_SIM_SPEED_FACTOR=20 make px4_sitl jmavsim`.

```
pxh> mavlink status

instance #0:
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 9.399 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: UDP (14570, remote port: 14550)

instance #1:
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 16.743 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14580, remote port: 14540)
```

```
pxh>  listener telemetry_status

TOPIC: telemetry_status instance 0 #1
 telemetry_status_s
	timestamp: 966480000  (1.588000 seconds ago)
	heartbeat_time: 0
	data_rate: 4000000.0000
	rate_multiplier: 1.0000
	rate_rx: 0.0000
	rate_tx: 16.7450
	rate_txerr: 0.0000
	remote_system_id: 0
	remote_component_id: 0
	remote_type: 0
	remote_system_status: 0
	type: 0
	mode: 2
	flow_control: False
	forwarding: False
	mavlink_v2: False
	ftp: True
	streams: 43
```

```
pxh>  uorb top -1

update: 1s, num topics: 53
TOPIC NAME                        INST #SUB #MSG #LOST #QSIZE
geofence_result                      0    1    4     0 1
vehicle_global_position              0    5  115   398 1
ekf_gps_drift                        0    0   10     0 1
telemetry_status                     0    2    2     0 3
vehicle_rates_setpoint               0    3  194   156 1
vehicle_attitude                     0    7  231  1029 1
ekf2_innovations                     0    1  115   107 1
sensor_bias                          0    4  115   291 1
vehicle_odometry                     0    2  115   207 1
battery_status                       0    6   83   273 1
ekf2_timestamps                      0    1  231    51 1
multirotor_motor_limits              0    2  250    53 1
rate_ctrl_status                     0    1  250   217 1
vehicle_magnetometer                 0    4  232   547 1
vehicle_air_data                     0    4  232   510 1
airspeed                             0    4  231   500 1
vehicle_land_detected                0    8    1     0 1
vehicle_local_position_groundtruth   0    1  125   110 1
vehicle_global_position_groundtruth  0    1  125   110 1
vehicle_attitude_groundtruth         0    1  125     0 1
differential_pressure                0    3  250   551 1
actuator_controls_0                  0    5  250   676 1
sensor_combined                      0    4  233   550 1
actuator_outputs                     0    4  250   676 1
vehicle_local_position               0    7  115   490 1
vehicle_status_flags                 0    1    1     0 1
commander_state                      0    0    1     0 1
actuator_armed                       0    6    1     0 1
vehicle_status                       0    9    1     0 1
vehicle_control_mode                 0    8    1     0 1
estimator_status                     0    4  115   515 1
sensor_preflight                     0    1  233   226 1
vehicle_gps_position                 0    5   10     9 1
sensor_baro                          0    1  250    18 1
sensor_mag                           0    2  250   218 1
sensor_gyro                          0    3  250   217 1
sensor_accel                         0    2  250   220 1
```

lamping7, 2019-06-20:
I saw this randomly yesterday when doing some multi-uav sims with ROS. I didn't see this topic before-hand. So, I don't have more than confirming that I saw "ERROR [uorb] telemetry_status advertise failed." I troubleshot a few things during that time, so hard to say more considering the environment changes I made during that time. I didn't touch speed factor.

julianoes, 2019-07-01:
I have seen it at times without speed up as well.

julianoes, 2019-10-01:
Doesn't seem to be happening anymore.

julianoes, 2019-10-09:
This is happening again, when using Gazebo.

dagar, 2020-01-07:
This has been fixed.

closed 2020-01-07 by dagar