PX4/PX4-Autopilot #12267: UAV is don't follwing the waypoints in mission plane

sampathkumarch, 2019-06-14:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

julianoes, 2019-06-14:
We can't help like that. You would have to give some detail as laid out in the issue template and at least add a logfile.

sampathkumarch, 2019-06-14:
https://review.px4.io/plot_app?log=e74f3807-eb85-4861-85f4-7ee5714dcf8f

Given waypoints

julianoes, 2019-06-14:
Could you add the mission plan file that you used in QGC if you have it?

sampathkumarch, 2019-06-14:
Given mission plane waypoints

sampathkumarch, 2019-06-14:
please give me reply any body

LorenzMeier, 2019-06-15:
You need to attach the mission plan file in Github. You have replied to the email, but that doesn't attach it in the web view - please use that.

sampathkumarch, 2019-06-15:
sir ,thank for a given a reply
 this the pic of mission plane
https://drive.google.com/file/d/1Zd9IMk7e07cHIWkC-oapEv4A8YTR9WOJ/view?usp=sharing

julianoes, 2019-09-17:
>  this the pic of mission plane

I have no time to translate a picture to a .mission plan file from QGC, sorry. If you want an answer you need to make it very easy for us to reproduce, and that's by adding a .mission file.