PX4/PX4-Autopilot #12268: UAV is don't following the waypoints in mission plane

sampathkumarch, 2019-06-14:
**Describe the bug**

am using pixhawk v5 . pixhawk was integrated with standred plane but the plane was no rudder direction ( airframe reference standre plane ). all sensor calibration are done then after i flyed in manual and stabilize mode . then after i given way points for mission plane so when am start the mission then given little bit of movement the plane was started then i throw plane but the plane was not going proper waypoint it going wrong direction . that's why am change the mode into stabilize mode . my point is why it's going wrong waypoints . rudder problem or parameters issue or anything else please help me

**To Reproduce**
**Steps to reproduce the behavior**

1. all sensor calibrated gyro, mag ,airspeed ,comp. level, radio
2. intially fly with stabilized and manual mode
3. Uploaded mission (attach QGC mission file)
4 .Took off
5.error is go the wrong waypoints
6. Expected behavior

**Log Files and Screenshots**

https://review.px4.io/plot_app?log=e74f3807-eb85-4861-85f4-7ee5714dcf8f

julianoes, 2019-06-14:
Duplicate of #12267.

closed 2019-06-14 by julianoes

sampathkumarch, 2019-06-14:
this is the am given mission plane