PX4/PX4-Autopilot #12278: Lidar Lite V3 newer than 2016 is not recognized by Pixhawk 2.1

ar1040, 2019-06-14:
The Lidar Lite V3 rangefinder is an optional rangefinder to be used with the Pixhawk 2.1. It can be enabled by the parameter EKF2_RNG_AID=1 and the Lidar Lite V3 can be initialized by the parameter SENS_EN_LL40LS. I use the I2C protocol.

I currently have 3-4 Lidar Lite V3's that are dated 2016 on the label. These Lidars work every time. The tests I have ran:

- Visual: Noticing the craft holds altitude
- The Analyze tab in QGroundControl: In QGroundControl, look in the top left corner, click widgets, then analyze. Type distance_sensor_current in the filter box. If the Lidar is working it will display a parameter that can be clicked and data can be viewed. (Note: the Lidar must be connected to the Pixhawk correctly, and both the Pixhawk and the Lidar have to be powered. I will add links to the correct way to wire this at the end)
- Arduino test:  Connected the Lidar to the arduino using the provided documentation.  https://learn.sparkfun.com/tutorials/lidar-lite-v3-hookup-guide/all

The 2016 Lidars will pass all of the tests described above. The 2018-2019 Lidars will only pass the arduino test.

I decided to go straight to the source by contacting Garmin. Garmin's response was, "The newer units use a newer hardware design that uses a faster FPGA.  The intent was for it to function as a drop-in replacement for the original, but there are some differences.  We have made updates to the https://github.com/garmin/LIDARLite_Arduino_Library that are intended to account for the differences."

Do you guys plan to support this new change? When could I expect the change to be implemented if so? Is there anything I can do in the meantime to fix this? Any and all help is appreciated.

ar1040, 2019-06-14:
[LLv3-I2C-Guidelines.pdf](https://github.com/PX4/Firmware/files/3292281/LLv3-I2C-Guidelines.pdf)

mcsauder, 2019-07-05:
@ar1040 , I will put some time into this issue in the next week or so and report back here as I make progress.

mcsauder, 2019-07-10:
@ar1040, my hardware is a V3HP and fortunately/unfortunately it works as expected with current PX4:master firmware flashed to a pixhawk 4 (fmu-v5). Where did you obtain your newer units and are they the V3 or V3HP?

ar1040, 2019-07-11:
@mcsauder, my hardware is a V3. I bought these units from robotshop.com, but they are original Garmins. I can use the PWM protocol and it works, but it only reads 12 meters max, versus 17 meters using I2C so I would prefer to have the extra 5 meters of distance. 

I think it is also worth mentioning I am using a Pixhawk Cube and not the Pixhawk 4. I am also using the 1.8.2 f13bbac firmware version. I would like to swap to the 1.9.2 firmware but there are still issues with takeoff and attitude that conflicts with my software, but this is a different issue in itself.

mcsauder, 2019-07-12:
Hi @ar1040 .  This might be very difficult to resolve if we cannot move you to the current release firmware; there are many bugfixes and code changes in the past year that might complicate resolving the issue or render it irrelevant. Could you raise a new issue to deal with the reason you cannot update to new firmware?

ar1040, 2019-07-15:
@mcsauder  I would like to use the newest firmware version due to the other fixes. I will open a new issue regarding the firmware issues. If the changes for the Lidar are pushed to the latest firmware version, it would still be very helpful, as I plan to switch our units to the latest firmware.

mcsauder, 2019-07-17:
@ar1040 , I've flashed v1.8.2 and and see an issue that might be the same as yours with my fmu-v5 hardware, the ll40ls i2c driver fails to initialize properly in the `LidarLiteI2C::probe()` call.

So, I think you have two options: 1) Upgrade to v1.9.2, or 2) backport the current driver `probe()` method to 1.8.2 and keep running with what you have. 

I have backported to v1.8.2 by copying the current PX4:master branch `probe()` method and a few #defines.  

You'll need to add these #defines to `LidarLiteI2C.h`
```

#define LL40LS_UNIT_ID_HIGH 0x16
#define LL40LS_UNIT_ID_LOW 0x17
/* Device limits */
#define LL40LS_MIN_DISTANCE (0.05f)
#define LL40LS_MAX_DISTANCE_V3 (35.00f)
#define LL40LS_MAX_DISTANCE_V1 (25.00f)
```
and this variable in the class:
```
uint16_t	_unit_id;
```

and copy/paste the current `probe()` method from [here](https://github.com/PX4/Firmware/blob/51d4239eee63d2eabf3e1980e04c2b4718984ceb/src/drivers/distance_sensor/ll40ls/LidarLiteI2C.cpp#L172).

As this appears to not be an issue with v1.9.2 and newer code, we can probably close out this issue.  Let us know your thoughts.

ar1040, 2019-07-18:
@mcsauder , I updated another one of my Pixhawk Cubes to v1.9.2 but I still cannot get a distance sensor reading. I normally test by opening QGC and clicking the widgets tab on the top left, then analyze. I then type "distance" in the search bar of the analyze window. If the sensor is recognized, it gives me 8 (I think) data sets to view. I choose the distance_sensor_current data set and view it to make sure it is reading the distance correctly.

However, even after I updated to v1.9.2 I could not get my Lidar Lite V3 to register.

mcsauder, 2019-08-01:
@ar1040 , can you test again with today's current PX4:master branch?

ar1040, 2019-08-01:
@mcsauder , I pulled the latest master via QGC (dropdown and selected master). The firmware shows up as 1.9.0dev. I am not sure if I am pulling the correct firmware version. But I tested it anyway. Still, only my 2016 lidar works and none of the 2019s work with I2C (I tried 5 that I had on hand). If I am not testing the correct firmware please let me know.

larsje39, 2019-08-04:
Hello, I am having the same issue (also with the master branch). One thing I am noticing is that if I boot using an older Lidar I see the data measurements, and if I leave the drone on and connect the newer lidar it works as well. However, when I do a boot using the newer lidar it does not work.
With kind regards,
Lars

mcsauder, 2019-08-05:
@ar1040 ,
> @mcsauder, yes my is working now, but only if I run the ll40ls start command in the Mavlink Console, otherwise it never shows in the analyze window.

I've tried mine on the bench connected to I2C A and it starts correctly at boot with `SENS_EN_LL40LS` set to I2C (2).  Which port are you connected to so that I can try to get as close to your configuration as I can?

ar1040, 2019-08-05:
@mcsauder 

> I've tried mine on the bench connected to I2C A and it starts correctly at boot with SENS_EN_LL40LS set to I2C (2). Which port are you connected to so that I can try to get as close to your configuration as I can?<

I am using the only I2C port available on the Pixhawk 2.1 Cube that I am aware of.

HenryHuYu, 2019-08-14:
> @ar1040 ,
>
>
> I've tried mine on the bench connected to I2C A and it starts correctly at boot with `SENS_EN_LL40LS` set to I2C (2). Which port are you connected to so that I can try to get as close to [...]

This is actually common problem as I am also met with this. The follows is what Garmin replied in email: 
If the manufacture date on the sensor is after 08/2018, you will need to update your drivers in order to make it work properly. 
You can refer to this support topic  article for more info https://support.garmin.com/en-US/?faq=BI0Aphz9cx5OXvG3fxnux6&partNumber=010-01722-00&searchQuery=i2c&tab=topics&textPage=1

Is this solved yet?

mcsauder, 2019-08-14:
I'm refactoring the driver to try to solve this...  Once I get a PR in, would you all help me test?

mcsauder, 2019-08-14:
No, this isn't solved yet.  Please feel free to submit a PR or see also #12695

mcsauder, 2019-09-12:
Hi @HenryHuYu and @ar1040 , would you give current master branch a run and see if this issue is solved?  Thanks!

mcsauder, 2019-09-17:
I think this is fixed now.... or at least I hope it is.  Please re-open if it is still an issue!

ar1040, 2019-09-23:
@mcsauder, I apologize for the late response. I did pull the master and I am still having the same problem as I can not get data from the lidar unless I start it with the "ll40ls start" command in the mavlink console.

[image]

julianoes, 2020-01-30:
Looks like a potential fix has been merged. Closing, please re-open if the issue persists.

closed 2020-01-30 by julianoes

yuthikasagarage, 2020-07-16:
I purchased a lidarlite which is manufactured 2019/01, i'm getting the same issue on 1.10.1
It doesn't show up and also even whether i start the driver it is giving me the following error
[
[image]
](url)

yuthikasagarage, 2020-07-16:
it seems my lidarlite has some problem, even in arduino it only gets nack message, not the same issue