PX4/PX4-Autopilot #12292: Can't change UAV-Type in Multi-UAV SITL Simulation

tensarflow, 2019-06-16:
**Describe the bug**
Multi-UAV SITL Simulation with IRIS works fine. But I think there is a problem for the standard plane. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set up SITL with Gazebo Environment for multiple UAVs as described [here](https://dev.px4.io/en/simulation/multi-vehicle-simulation.html) .
2. Change 
`<arg name="vehicle" default="iris"/>` 
to 
`<arg name="vehicle" default="plane"/>` 
in [multi_uav_mavros_sitl.launch](https://github.com/PX4/Firmware/blob/master/launch/multi_uav_mavros_sitl.launch).
3. Run `roslaunch px4 multi_uav_mavros_sitl.launch`.
4. See error

**Expected behavior**
I would like to have multiple fixed wing planes in my gezebo simulation

**Log Files and Screenshots**
This is the error I get:
```
ensasr@ensarspc:~/src/Firmware$ roslaunch px4 multi_uav_mavros_sitl.launch
... logging to /home/ensasr/.ros/log/26e55dba-9033-11e9-9301-5800e3df8d01/roslaunch-ensarspc-11643.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
No such file or directory: /home/ensasr/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/plane_base.xacro
RLException: while processing /home/ensasr/src/Firmware/launch/single_vehicle_spawn.launch:
Invalid <param> tag: Cannot load command parameter [rotors_description]: command [/opt/ros/melodic/lib/xacro/xacro /home/ensasr/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/plane_base.xacro rotors_description_dir:=/home/ensasr/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 mavlink_tcp_port:=4560 --inorder] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="rotors_description"/>
The traceback for the exception was written to the log file

```
And this is the log I get:
```
[roslaunch][INFO] 2019-06-16 14:34:58,215: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2019-06-16 14:34:58,221: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2019-06-16 14:34:58,221: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'px4', 'multi_uav_mavros_sitl.launch']
[roslaunch][INFO] 2019-06-16 14:34:58,221: roslaunch env is {'LC_NUMERIC': 'en_US.UTF-8', 'ROS_DISTRO': 'melodic', '_LXSESSION_PID': '1151', 'ROS_LOG_FILENAME': '/home/ensasr/.ros/log/26e55dba-9033-11e9-9301-5800e3df8d01/roslaunch-ensarspc-11643.log', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/ensasr', 'XDG_DATA_HOME': '/home/ensasr/.local/share', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'PKG_CONFIG_PATH': '/home/ensasr/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'XDG_RUNTIME_DIR': '/run/user/1000', 'XDG_CURRENT_DESKTOP': 'LXDE', 'LC_PAPER': 'de_DE.UTF-8', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/ensasr/catkin_ws/devel/share/common-lisp', 'LOGNAME': 'ensasr', 'USER': 'ensasr', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-LXDE:/etc/xdg', 'PATH': '/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'XDG_VTNR': '7', 'HOME': '/home/ensasr', 'CMAKE_PREFIX_PATH': '/home/ensasr/catkin_ws/devel:/opt/ros/melodic', 'DISPLAY': ':0.0', 'SSH_AGENT_PID': '1214', 'LANG': 'en_US.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'XAUTHORITY': '/home/ensasr/.Xauthority', 'LANGUAGE': 'en_US', 'LC_MEASUREMENT': 'de_DE.UTF-8', 'MANDATORY_PATH': '/usr/share/gconf/LXDE.mandatory.path', 'LD_LIBRARY_PATH': '/home/ensasr/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo:/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo:/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo', 'QT_ACCESSIBILITY': '1', 'ROS_MASTER_URI': 'http://localhost:11311', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'ROS_PYTHON_VERSION': '2', 'XDG_CONFIG_HOME': '/home/ensasr/.config', 'SAL_USE_VCLPLUGIN': 'gtk', 'ROS_VERSION': '1', 'LC_ADDRESS': 'de_DE.UTF-8', 'PYTHONPATH': '/home/ensasr/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/tmp/ssh-Xca4AwDfuguz/agent.1151', 'SHLVL': '1', 'VTE_VERSION': '5202', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'GDMSESSION': 'LXDE', 'LC_TIME': 'de_DE.UTF-8', 'XDG_DATA_DIRS': '/usr/local/share:/usr/share:/usr/share/gdm:/var/lib/menu-xdg:/usr/local/share/:/usr/share/:/usr/share/gdm/:/var/lib/menu-xdg/', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'ROS_PACKAGE_PATH': '/home/ensasr/catkin_ws/src/mavlink:/home/ensasr/catkin_ws/src/mavros/libmavconn:/home/ensasr/catkin_ws/src/mavros/mavros_msgs:/home/ensasr/catkin_ws/src/mavros/mavros:/home/ensasr/catkin_ws/src/mavros/mavros_extras:/home/ensasr/catkin_ws/src/pysdf:/home/ensasr/catkin_ws/src/mavros/test_mavros:/opt/ros/melodic/share:/home/ensasr/src/Firmware:/home/ensasr/src/Firmware/Tools/sitl_gazebo', 'GAZEBO_MODEL_PATH': ':/home/ensasr/src/Firmware/Tools/sitl_gazebo/models:/home/ensasr/src/Firmware/Tools/sitl_gazebo/models:/home/ensasr/src/Firmware/Tools/sitl_gazebo/models', 'XDG_SESSION_ID': 'c2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/opt/ros/melodic/bin/roslaunch', 'GTK_MODULES': 'gail:atk-bridge', 'LC_IDENTIFICATION': 'de_DE.UTF-8', 'DESKTOP_SESSION': 'LXDE', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'DEFAULTS_PATH': '/usr/share/gconf/LXDE.default.path', 'XDG_SESSION_TYPE': 'x11', 'OLDPWD': '/home/ensasr/catkin_ws', 'GDM_LANG': 'en_US', 'LC_TELEPHONE': 'de_DE.UTF-8', 'GAZEBO_PLUGIN_PATH': ':/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo:/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo:/home/ensasr/src/Firmware/build/px4_sitl_default/build_gazebo', 'LC_MONETARY': 'de_DE.UTF-8', 'PWD': '/home/ensasr/src/Firmware', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'COLORTERM': 'truecolor', 'LC_NAME': 'de_DE.UTF-8', 'XDG_MENU_PREFIX': 'lxde-', 'XDG_SESSION_DESKTOP': 'LXDE', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'XDG_SEAT': 'seat0'}
[roslaunch][INFO] 2019-06-16 14:34:58,221: starting in server mode
[roslaunch.parent][INFO] 2019-06-16 14:34:58,221: starting roslaunch parent run
[roslaunch][INFO] 2019-06-16 14:34:58,221: loading roscore config file /opt/ros/melodic/etc/ros/roscore.xml
[roslaunch][INFO] 2019-06-16 14:34:58,615: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2019-06-16 14:34:58,616: loading config file /home/ensasr/src/Firmware/launch/multi_uav_mavros_sitl.launch
[roslaunch][INFO] 2019-06-16 14:34:58,653: Added node of type [gazebo_ros/gzserver] in namespace [/]
[roslaunch][INFO] 2019-06-16 14:34:58,654: Added node of type [gazebo_ros/gzclient] in namespace [/]
[roslaunch][ERROR] 2019-06-16 14:34:58,730: RLException: while processing /home/ensasr/src/Firmware/launch/single_vehicle_spawn.launch:
[... cut ...]
The [posix_sitl.launch](https://github.com/PX4/Firmware/blob/master/launch/posix_sitl.launch) for single UAV's are super-compatible with the `SDF`s.

I tried to adjust `posix_sitl.launch` for the `SDF`s. Unfortunately I don't have much experience with these XML-things and then there are these weird references to other launch files and parameters and stuff. :smile: So it would be really great if someone with experience could handle this very quick. In the meantime I'm going to try to fix that too and maybe create a pull request if I think it works, since I don't have much to do after my fresh graduation hehe
Cheers!

tensarflow, 2019-06-16:
I'm starting to understand... The SDF files contain the UDP port binding information of the simulation. Since they need to be unique for each UAV you can't use the same SDF for multiple simulations. That's not good coding :smile:
I'm thinking of just making multiple SDF files and trying to launch them from one launch file... Still on it :)

lamping7, 2019-06-16:
This is not a firmware issue. It is integration related. You should post a topic on http://discuss.px4.io for help.

closed 2019-06-16 by lamping7