PX4/PX4-Autopilot #12294: GPS takes way too long to lock or not locking at all

HVKHVK, 2019-06-17:
**Describe the bug**
At v1.9.0 GPS takes way too long to lock or not locking at all. I can see that 7 satellites are active on Qgroun but it is still not locking. Sometimes it locks suddenly but many times I just give up. Because of not have the ability to use position and mission modes, I downgraded the firmware to v1.8.2. 

**To Reproduce**
After enabling power to the PX4, it happens every single time.

**Expected behavior**
At v1.8.2 it took a maximum of 2 minutes to have a decent lock. So I was expecting the same on v1.9.0.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=813ee3c2-4b82-4b9c-89c8-02c76961ccaa

RomanBapst, 2019-06-17:
@HVKHVK have a look at this
https://docs.px4.io/en/advanced_config/tuning_the_ecl_ekf.html#gps-performance-requirements

RomanBapst, 2019-06-17:
@HVKHVK You can adjust those parameters until the checks pass. Note however, that the default values are not necessarily too low but instead you might have some interference or there might be some particularity on your setup which requires the parameters to be changed.

HVKHVK, 2019-06-17:
I did, but at same GPS location with the same device, v1.8.2 and v1.9.0 act very differently. I also edited some of those parameters too but still the same. I will try to find the optimum than I will report those.

RomanBapst, 2019-06-17:
@HVKHVK You can also disable gps checks using EKF2_GPS_CHECK and then add one by one to see which one might be the offending one.
I'm going to close this as I don't think it's a bug in the code. You can also use the forums if you have more questions.
And if you do think that you've found a bug in the code then please re-open.

closed 2019-06-17 by RomanBapst