PX4/PX4-Autopilot #12307: RC loss behavior

RyanHurst, 2019-06-19:
**Describe the bug**
Drone lost communication and then descended to ground.

**To Reproduce**
Steps to reproduce the behavior:
Fly drone in Position mode and lose communication with it. I don't know how to reproduce this consistently.

**Expected behavior**
I would expect the drone to continue to do what it was doing before because NAV_RCL_ACT  was set to 0.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=d97a8a6f-94a6-42fa-abf0-0dff97ba8102

**Drone:**
- Teal One

**Additional context**
PX4 1.9.0

dakejahl, 2019-06-21:
[image]

dakejahl, 2019-06-22:
Take a look at the `failsafe` function in `mc_pos_control_main`. Specifically line `563`

I'm guessing this failsafe is getting hit. If you look at your XYZ setpoints in your log, you'll see they abruptly stop as soon as RC is lost. If nothing is controlling the drone (no RC, no setpoints) really the only reasonable thing the drone can do is land.

You should verify what I am saying by testing though, this is just a guess.

tops4u, 2019-06-24:
I think the question is, what would be the expected behaviour? If the Drone is flying with 5km/h away from the Pilot would the expected behaviour be, that it should continue doing this until the Battery Failsafe kicks in? In PosCtrl Mode Landing or RTL is probably the only sensible thing to do. If you want it to stay where it is then you should configure the Param to 1. If in Mission Mode a Param Value of 0 would most probably continue the Mission.

RyanHurst, 2019-06-24:
I had some time to do some more testing with this in simulation.
In position mode: The drone will continue moving in the last direction it was going, but it will "descend with land speed"
In mission mode: The drone will finish the mission

My issue is with the position mode behavior.
Things I could see the drone doing:
1. Shift into land mode
2. Stay in position mode, but update the setpoints so that the drone stops moving
3. Keep moving at the same speed, and in the same direction with the same yaw

the 3rd option doesn't sound very useful or safe, but maybe 1 or 2?

dakejahl, 2019-06-24:
> 1. Shift into land mode
I think this is what the design intent was, but if it is keeping its velocity then this seems like a bug, as the vehicle should never try to land while moving
> 2. Stay in position mode, but update the setpoints so that the drone stops moving
This is what NAV_RCL_ACT == 1 effectively accomplishes. Position mode is a manual mode, so it does not make sense for the vehicle to issue itself setpoints. You should just use NAV_RCL_ACT == 1 and then when you want to start sending control setpoints again issue a mode change back to position mode.

julianoes, 2019-10-03:
So I tried to reproduce this. I've set the params as follows:
`NAV_RCL_ACT` to `3` -> Land
`NAV_DLL_ACT` to `3` -> Land

- When I fly in POSCTL with QGC connected and then unplug the joystick it correctly goes into Land mode and lands. (The confusing part is that the drone goes back to the location where RC was lost so it actually moves back a bit while landing.)
- When I fly a mission and unplug the joystick it finishes the mission and then waits at the last waypoint. This is because you can still send it back using QGC (datalink is still ok). If I close QGC it goes to land once it considers the datalink lost.

Now let's see what happens with failsafe action disable:
`NAV_RCL_ACT` to `0` -> Disabled
`NAV_DLL_ACT` to `0` -> Disabled

- When I fly in POSCTL with QGC connected and then unplug the joystick it stays in Position mode but it also says: `WARN  [mc_pos_control] Failsafe: Descend with land-speed.` **which seems wrong!**
- When I fly in MISSION and then unplug the joystick it completes the mission and then hovers there forever, presumably until the battery failsafe kicks in. If QGC is shut down and the data link is lost, it still keeps hovering there.

julianoes, 2019-10-03:
@MaEtUgR it's correct what @dakejahl wrote above.

We end up in the flight task failsafe here:
https://github.com/PX4/Firmware/blob/5f040fe24c1e488c6b7aadefa37c168bfa96b190/src/modules/mc_pos_control/mc_pos_control_main.cpp#L592

But actually we should just stay at the last position control position. Do you understand what's going on here?

MaEtUgR, 2019-10-08:
@julianoes Sorry, I've seen it too late.

Yes, I agree with
> Position mode is a manual mode, so it does not make sense for the vehicle to issue itself setpoints.

The problem is in my opinion the following:
The failsafe handling in the position controller is a stub and far from finished. It was started as the future way of reporting these cases but since the feedback to existing failsafe handling is a huge pain it's so far unfinished. I currently consider it the last resort emergency for cases not properly handeled by commander.

Now if you set the configuration to not react at all and you have RC loss in position mode and commander doesn't do anything and the vehicle stays in position mode and runs the flight task executing it I consider it unhandled and unsafe because the setpoint generation relies on RC data and it's not available anymore. The outcome is therefore undefined. That's also why the flight task reports an error and the position controller executes last resort.

Bottom line TL;DR according to my opinion there must be a reaction to RC loss during manual flight. And I agree that immediate descend speed is not the best solution. Let me check #13125

MaEtUgR, 2019-10-08:
I made a suggestion in #13130 . Could you check if it fixes the issue for you?

MaEtUgR, 2019-10-14:
Might not have been fixing all the problems that were stated in this issue, therefor reopening.

julianoes, 2019-10-22:
I think we can close this issue and then revisit all the failsafes scenarious when I get to working on that in general.

PR #13130 (2019-10-14): Position Controller Failsafe Improvment
**Describe problem solved by the proposed pull request**
Presumably fixes #12307 
@julianoes @RyanHurst Can you please check if that produces the desired behavior?
In my quick SITL tests it worked fine.

**Test data / coverage**
SITL testing enabling disabling virtual joystick in QGC to simulate RC loss.

**Describe your preferred solution**
Since commander should still handle all failsafes we should only run into this case as last resort to not crash. If all failsafe actions are disabled but data is missing e.g. RC loss action disabled but flying in man [... cut]
closed 2019-10-22 by julianoes