PX4/PX4-Autopilot #12309: Mavlink PLAY_TUNE message does not play tune

wluker, 2019-06-19:
**Describe the bug**
Sending a tune string "MFT200CDEFGAB" via a PLAY_TUNE mavlink message does not produce any sound on the drone. 

From the fix for #12059 I can produce sound with "tune_control play -m "MFT200CDEFGAB" ".

The PLAY_TUNE mavlink handler is implemented in https://github.com/PX4/Firmware/blob/4831a4b0cfde499eb1a20257cc46b130e6554e3a/src/modules/mavlink/mavlink_receiver.cpp#L1627 and I am sending the correct message because this triggers, but it doesn't produce sound.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. send PLAY_TUNE mavlink message with "MFT200CDEFGAB" in PLAY_TUNE.tune
3. Listen for sound

**Expected behavior**
A tune should be played.

**Drone (please complete the following information):**
- Pixhawk2 with latest firmware from git

mcsauder, 2019-09-18:
Hi @wluker , if I issue the command `tune_control play -m "MFT200CDEFGAB" ` from the mavlink console in QGC the tune plays correctly for me on fmu-v2 and fmu-v5.

I think we can close this issue, but if I am not understanding your issue or this is still an issue for you would you please re-open?  Thanks!

wluker, 2019-09-18:
Hi @mcsauder 

Thanks for looking into this. This issue is not in `tune_control play -m "MFT200CDEFGAB"` because that plays for me as well. The issue is that you can't trigger any tune via the mavlink message `PLAY_TUNE`. https://mavlink.io/en/messages/common.html#PLAY_TUNE

I was trying to send that message via an offboard controller, but no tones were being played even though I could tell PX4 was receiving the message. Maybe I am misunderstanding the intended use of the `PLAY_TUNE` message.

julianoes, 2020-01-30:
Fix coming in #13158.

julianoes, 2020-04-29:
Fixed in #13158.

closed 2020-04-29 by julianoes

mrpollo, 2020-04-29:
In your face @stale bot! 

Thanks @julianoes 🎵🎶🎵🎶