PX4/PX4-Autopilot #12313: Here2 not being detected with PX4 firmware

donbreum, 2019-06-21:
I recently bought a Pixhawk Cube combo kit (Pixhawk 2.1 Cube with Here2 kit). I have previous experience with this kit and never experienced issues

Now I am not able to detect the Here2 GNSS receiver if I am using PX4 firmware (using newest stable build 1.9). If I flash newest ArduCopter firmware, the GNSS is detected and is working fine

The new Here2 was supplied with the CAN cable attached and I changed this to the serial cable for GPS1 serial port. Also, I've flicked the Serial+I2C switch inside the Here2 kit to Serial+I2C (according to http://www.hex.aero/wp-content/uploads/2019/03/Here2-CAN-Instruction.pdf) 

I've tested with a Here2 GNSS which I've bought last year (this one was supplied with the GPS1 port cable). This works perfectly. 

Can anyone help me with the next step to solve this issue? It seems like a firmware problem since it works with ArduCopter firmware (but I need to use PX4 flight stack)

donbreum, 2019-06-25:
solved. problem was not related to px4 firmware, but the Here2 firmware

closed 2019-06-25 by donbreum

loulouagr, 2019-09-05:
hi @donbreum !
I have the same issue. Can you tell me how you manage to solve it ?
I have a dropix with here2 GNSS. It is in Serial + i2c and works fine with ardupilot, but is not detected by px4 flight stack.

donbreum, 2019-09-05:
> hi @donbreum !
> I have the same issue. Can you tell me how you manage to solve it ?
> I have a dropix with here2 GNSS. It is in Serial + i2c and works fine with ardupilot, but is not detected by px4 flight stack.

Actually I solved it by using the guide for updating bootloader. But i did this using the Pixhawk 2.1 Cube. I'm not sure if it's possible to do this using CAN bus with the dropix. Anyways i did the following myself:

I updated the bootloader on the Here2 and update the firmware. I’ve followed the instructions found here: https://docs.cubepilot.org/user-guides/here-2/here-2-instruction . 
 
Afterwards I switched the pin back to I2C and used the serial cable and connected the Here2 to the GPS 1 port on the pixhawk 2.1 module