PX4/PX4-Autopilot #12314: PX4 Models crash Gazebo

capitalPSA, 2019-06-21:
**Bug**
When inserting the 3DR Iris model into Gazebo, the error "Aborted (core dumped" occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch PX4 SITL ("no_sim=1 make px4_sitl_default gazebo")
2. Source the necessary directories (source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default)
3. Launch Gazebo (roslaunch gazebo_ros shapes_world.launch)
4. Insert the 3DR Iris model into the environment.
5. Error:
Aborted (core dumped)
[gazebo-2] process has died [pid 27778, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode worlds/shapes.world __name:=gazebo __log:=/home/USER/.ros/log/166013cc-9452-11e9-87a5-7085c2c3fe67/gazebo-2.log].
log file: /home/USER/.ros/log/166013cc-9452-11e9-87a5-7085c2c3fe67/gazebo-2*.log

**System Information**
Ubuntu: __18.04__
ROS: __Melodic__
Gazebo: __9.0.0__
PX4: __1.9.0__
GPU: __NVidia RTX 2070 (430.26)__
CUDA: __10.2__
Fresh Install: __Yes.__

julianoes, 2019-06-21:
Could you paste the output of the logfiles mentioned?

capitalPSA, 2019-06-23:
After some debugging, I am finding this error:
`0x00007fff04f66285 in
gazebo::GazeboMavlinkInterface::Load(boost::shared_ptr<gazebo::physics::Model>, std::shared_ptr<sdf::Element>) ()
    at /home/....../PX4ws/PX4/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so`

The other models, such as a ramp and shelves, can all be placed without any crashes. The models that require a MAVLINK interface (e.g. planes and multirotor platforms) are causing an `Aborted (core dumped)` with an error code of 134. 

I will continue to look further for additional debug information / insights.

Jaeyoung-Lim, 2019-06-23:
Have you tried adding the model in the launchfile?

capitalPSA, 2019-06-23:
@Jaeyoung-Lim You are right. Running the `px4.launch` file will insert the `Iris` model into the world and there is no crash. 

Is this the new preferred way? Add the model into the launch file instead of manually inserting the model after the world is loaded in Gazebo?

The error looks to be localized to adding a model (that requires MAVlink) after the world is loaded. If a model is spawned using the launch file, there is no crash. Additional models can be manually added if the first model is added using the launch file.

closed 2019-07-12 by capitalPSA