PX4/PX4-Autopilot #12321: Commander reports "Not Disarming, Not in manual mode or landed yet"

RomanBapst, 2019-06-24:
When flying in position control mode full stick down with full yaw left the commander tries to disarm but obviously blocks it if the vehicle is not landed yet.
However, it sends the message "Not Disarming, Not in manual mode or landed yet" to the ground station.
Is this message really useful?

RomanBapst, 2019-06-24:
This is where it happens:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/Commander.cpp#L1932

FlavioTonelli, 2019-06-24:
In my opinion it is. 
The drone can not always recognize instantly it is actually landed even if it on ground. First attempt to disarm the drone can then sometime fail.
The message helps the user to get what it's going on.

MaEtUgR, 2019-06-24:
The "manual mode" part can be misleading. It should either say not disarming because not landed yet or be more verbose (which I think is not possible in these messages) saying something like "Disarming with vehicle not having detected a landing is only possible in Stabilized and Acro"?

julianoes, 2019-07-03:
I agree that we should change it to a better message and probably it makes sense to limit it maybe only show up every 10 seconds or so.

julianoes, 2019-10-10:
This has been fixed by @MaEtUgR in #12687, closing.

closed 2019-10-10 by julianoes