PX4/PX4-Autopilot #12323: Optitrack using vision - drone hits ceiling

telegek, 2019-06-24:
The drone (DJI Flame Wheel F330) goes always high during take-off and hits the ceiling, even though after take-off the throttle stick is all they way down. Tried different autopilot systems, Pixhawk 1 and MindPX, the issue is the same. From the log the velocity setpoint does not seem to follow the command.

The topic from **vrpn_client_node** is relayed to the **mavros/vision_pose/pose**, EKF2_AID_MASK is set to vision position fusion and vision yaw fusion. The EKF2_HGT_MODE is set to vision.

Same setup with F450 frame seems working perfect and flying in position mode.

Log file attached.

https://review.px4.io/plot_app?log=6c64a5ae-5aeb-4327-a68b-c6fc60d5b130

Any ideas? The help would be appreciated.

fabianschilling, 2019-07-03:
I ran into the same issue a while ago. Try to lower the vertical thrust that is required to hover with the  `MPC_THR_HOVER` parameter. In my case, 30% was a good value but it will depend on the weight of your vehicle.

telegek, 2019-07-14:
> I ran into the same issue a while ago. Try to lower the vertical thrust that is required to hover with the `MPC_THR_HOVER` parameter. In my case, 30% was a good value but it will depend on the weight of your vehicle.

Thanks. This solved an issue!

closed 2019-07-14 by telegek