PX4/PX4-Autopilot #12326: jmavsim GUI failed to launch

yuhaim, 2019-06-25:
**Describe the bug**
jmavsim GUI failed to launch

**To Reproduce**
Ubuntu 18.04 with openJDK 11 for default-jdk, default-jre
start jmavsim alone or start SITL

**Expected behavior**
jmavsim GUI popups and simulation begins.

**Log Files and Screenshots**
~/src/Firmware$ ./Tools/jmavsim_run.sh -l
Buildfile: /home/miller/src/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

~/src/Firmware$ make px4_sitl_default jmavsim
[0/1] cd /home/miller/src/Firmware/bui...er/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/miller/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/miller/src/Firmware
build_path: /home/miller/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/miller/src/Firmware/build/px4_sitl_default/bin/px4" "/home/miller/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/miller/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/miller/src/Firmware/ROMFS/px4fmu_common -> /home/miller/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
Buildfile: /home/miller/src/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Inconsistency detected by ld.so: dl-lookup.c: 111: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!

AlreadyUncle, 2019-07-29:
Hi, i'm having exactly the same problem. Did you find a solution ?

julianoes, 2019-07-29:
This should solve it:
https://github.com/PX4/jMAVSim/issues/96#issuecomment-500788800

closed 2019-07-29 by julianoes