PX4/PX4-Autopilot #12341: px4_fmu-v5 firmware v1.9.0 doesn't check for external I2C lis3mdl compass

dlwalter, 2019-06-26:
**Describe the bug**
v1.9.0 v5 firmware does not check for an I2C external LIS3MDL compass on Pixhawk 4 mini

**To Reproduce**
plug in an LIS3MDL compass - it won't show as connected. You can start it manually with "lis3mdl -X start"

**Expected behavior**
firmware should check for lis3mdl compass on external i2c as part of boot sequence

The following line should be added to Firmware/boards/px4/fmu-v5/init/rc.board_sensors

```
# Possible external compasses
ist8310 -C -b 1 start
ist8310 -C -b 2 start
hmc5883 -C -T -X start
qmc5883 -X start
**lis3mdl -X start**

```

dlwalter, 2019-07-01:
@dagar This is marked as closed.  Is 1.9.0 not supporting the LIS3MDL compass?

dagar, 2019-07-01:
If you can open a PR with that change and test it quickly I can get it into v1.9.2.

dlwalter, 2019-07-01:
https://github.com/PX4/Firmware/pull/12384

Done.

closed 2019-07-01 by dagar