PX4/PX4-Autopilot #12351: Add vehicle model module for converting wheel encoder counts into body velocities

ItsTimmy, 2019-06-27:
**Describe problem solved by the proposed feature**
Wheel encoder support is being worked on [here](https://github.com/ItsTimmy/Firmware/tree/pr-roboclaw). The end result of this drive will be a uORB topic that publishes raw encoder data. This data must be converted to velocities to be used by the EKF2 module, but this is not trivial: Different rovers can have different wheel configurations and different number and placement of encoders.

**Describe your preferred solution**
A new module should be created which handles different rover configurations, and is easily extensible so that it can be easily updated to handle new configurations.

avielm, 2021-01-24:
is there any work on this?

dagar, 2021-01-24:
> is there any work on this?

The Roboclaw driver in current PX4 publishes odometry data. https://github.com/PX4/PX4-Autopilot/blob/f571b29ae8152cc0b3d0f6bb7f160e3a44a8d9ed/src/drivers/roboclaw/RoboClaw.cpp#L223-L229

The next step is actually consuming it in the ekf2, including adding appropriate configurable physical quantities for the wheels.