PX4/PX4-Autopilot #12355: Secondary GPS doesn't works

rligocki, 2019-06-27:
**Describe the bug**
On 1.9.1 stable I tried to connect two HERE2 GPS modules because of redundancy. GPS modules were connected to GPS1 and GPS2 ports on Pixhawk 2.1. Then I run GPS driver using command: "gps start -d /dev/ttyS3 -e /dev/ttyS4". The first GPS works ok, secondary GPS has no response 

**To Reproduce**
Steps to reproduce the behavior:
1. Pixhawk 2.1 with stable 1.9.1
2. console command "gps stop"
3. console command "gps start -d /dev/ttyS3 -e /dev/ttyS4".
4. console command "gps status"

**Expected behavior**
There should be two sources of GPS. 

**Log Files and Screenshots**
[image]

**Additional context**
I also tried different UARTs (/dev/ttyS1, ttyS2, ttyS6). At screenshot you can see ttyS6.

julianoes, 2019-07-01:
Have you seen this docs? https://docs.px4.io/en/peripherals/serial_configuration.html

rligocki, 2019-07-08:
Yep, I read it a few times. It does not work. I found that in default.cmake of FMU3 is GPS2 label missing. 

SERIAL_PORTS
		GPS1:/dev/ttyS3
		GPS2:/dev/ttyS4
		TEL1:/dev/ttyS1
		TEL2:/dev/ttyS2
		#TEL4:/dev/ttyS6

I added GPS2 label, now it starts after configuration of GPS2 interface. No need to use console.

julianoes, 2019-07-08:
Could you point me to where in the code this is missing? Thanks.

rligocki, 2019-07-08:
./px4rootFolder/boards/px4/fmu-v3/default.cmake. This is the missing label for GPS2 connector, but a setup for secondary GPS should work with TEL2 port. The problem still remains. I also tried to change primary and secondary GPS, but still, secondary GPS does not communicate.

julianoes, 2019-07-08:
@dagar do you understnd this?

rligocki, 2019-07-27:
Update: Problem is somewhere in pixhawk2.1 and here 2 GPS module. I tried same start command of GPS driver and it worked with pixhawk 2.4.8 and Ublox M8N GPS modules. I thought that cable is damaged, but unfortunately, cable looks ok. Also checked that GPS is in I2C mode.

closed 2019-07-31 by rligocki

RamanaBotta, 2022-01-05:
Same issue with as well. i zedf9p GPS receiver board. when i tried using it as secondary GPS it worked fine. after some days it stopped working. My output is also same as https://user-images.githubusercontent.com/9701384/60304474-da875a80-9939-11e9-9b9a-527a3a38d44c.png . So I solved it by re flashed the firmware on zedf9p from Ublox website. then i connected to pixhawk and it worked.