PX4/PX4-Autopilot #12359: Rate setpoint dropout during transition

RomanBapst, 2019-06-28:
While flying a quadplane one v1.9 I noticed that there is always a 1 second gap in the vehicle_rates_setpoint topic at the beginning of the transition. Normally the fixed wing attitude controller should be publishing values. The rate controllers still seem to run.

[image]

In the figure above you can see the gap in the rate setpoint topic (top right).
At the beginning of the transition we switch from multicopter attitude controller to fixed wing attitude controller to produce the rate setpoint. However, i don't see the reason yet why it takes the fixed wing attitude controller 1 second to publish to the rates setpoint topic.

@dagar Any ideas?

Log: https://logs.px4.io/plot_app?log=5bca4650-96a8-4f92-8619-b9b1cadccf4a

dagar, 2019-07-01:
I'm not sure, but I guess I'm not surprised given the complexity of the independent logic in each separate controller for coordinating which portion is running wrt vtol state and type.

I'm sure there's a whack-a-mole solution to the bug here if someone wants to temporarily get their head around all the current code and timing long enough to see the problem, but how about something architecturally simpler?

Currently we have `fw_att_control` and `mc_att_control` both containing attitude + rate controllers. What if we split these both into separate modules, and then turned [vehicle_control_mode](https://github.com/PX4/Firmware/blob/master/msg/vehicle_control_mode.msg) into an MC/FW virtual topic that the VTOL code could manipulate independently to control the transition behavior?

dagar, 2019-07-01:
Independent of the VTOL architecture I've already starting splitting mc_att_control. https://github.com/PX4/Firmware/pull/12225

RomanBapst, 2019-09-27:
@sfuhrer I believe you fixed this lately. Can you confirm and close?

sfuhrer, 2019-09-27:
> @sfuhrer I believe you fixed this lately. Can you confirm and close?

yes fixed

closed 2019-09-27 by sfuhrer

RomanBapst, 2019-09-27:
@sfuhrer You the man! Thanks.