PX4/PX4-Autopilot #12360: non-standard baudrates are not supported on this platform

oasinic, 2019-06-28:
Hi
Use windows 10 and PX4.Windows.Cygwin.Toolchain.0.5 
fork master today
compile firmware use 
`$ make px4_fmu-v3_default`
try to upload
`$ make px4_fmu-v3_default upload`
and get error 
`[0/1] uploading px4
Loaded firmware for board id: 9,0 size: 1665660 bytes (80.05%), waiting for the bootloader...

non-standard baudrates are not supported on this platform -> could not check for FTDI device, assuming USB connection
Attempting reboot on /dev/ttyS2 with baudrate=57600...
If the board does not respond, check the connection to the Flight Controller
non-standard baudrates are not supported on this platform -> could not check for FTDI device, assuming USB connection`

px4 connect to pc via usb cable and in windows recognized as COM3

Thanks!

julianoes, 2019-10-01:
@MaEtUgR do you understand this?

MaEtUgR, 2019-10-21:
> non-standard baudrates are not supported on this platform -> could not check for FTDI device, assuming USB connection

Is a warning that always shows on Cygwin because pyserial cannot set certain baudrates that are used for checking FTDI corner cases and so it assumes you're flashing through USB. Nothing to worry about there.

The real problem you're facing is that you try to flash, it sees the connected pixhawk and sends it a reboot to bootloader command: `Attempting reboot on /dev/ttyS2 with baudrate=57600...`

But then apparently the bootloader does not get recognized afterwards.

@oasinic Could you check if there are other COM port devices showing up in your device manager?
Because the bootloader will get a different COM port when it's connecting to your machine and if there's any other device the upload script could grab and get stuck on it would explain the problem. In that case you might want to test if it solves the problem temporarily disabling the other COM port devices to see if it helps.