PX4/PX4-Autopilot #12361: List of default params that need an update

RomanBapst, 2019-06-28:
This is to track parameters which have a default that does not work well on practice or leads to inferior performance on pretty much every airframe.

Parameter name | default | suggested
------------------- | --------- | -------------
MPC_Z_VEL_I | 0.02| 0.25

LorenzMeier, 2019-06-28:
I think we need this in v1.9.2 already as we otherwise keep this issue too long.

DanielePettenuzzo, 2019-07-09:
SYS_FORCE_F7DC worries me a bit because the default is set to disable dcache. With dcache disabled we are very close to 100% cpu on fmu-v5 if we have a companion link, run uavcan or run the avoidance interface (many users could be hit by this). I just think that currently the risk of crashing due to a maxed out cpu is higher than actually hitting the dcache errata.

dagar, 2019-07-10:
> SYS_FORCE_F7DC worries me a bit because the default is set to disable dcache. With dcache disabled we are very close to 100% cpu on fmu-v5 if we have a companion link, run uavcan or run the [...]

@DanielePettenuzzo I'm pushing for a proper fix. https://github.com/PX4/Firmware/pull/12435

dagar, 2019-07-11:
@RomanBapst more updates coming or should we start making these changes (for v1.9.3 at this point)?

bresch, 2019-07-12:
We're using `MC_POS_MODE 3` (jerk-limited smoothing) since at least January on all our vehicles and I think it's the right time to change it to the upstream default. It will also help to receive more feedback from the user and the test team to improve it further.

fabianschilling, 2019-07-12:
The default `EKF2_EV_DELAY` of 175 ms seems rather high for flight with motion capture systems. A value around 50-100 ms could be a better default.

julianoes, 2019-07-15:
> I think we need this in v1.9.2 already as we otherwise keep this issue too long.

No, v1.9.2 is a patch release to fix critical bugs. I don't think people are expecting tuning changes. v1.10 is only 1.5 months away anyway.

LorenzMeier, 2019-07-15:
@julianoes My comment was in relation to a different list you’re looking at now. However, we need to talk through a couple of these parameters - a broken configuration is as bad as a broken piece of code. What is clear is that the core dev team has been flying different defaults than were released, which makes this a 1.9 release bug. We should treat this as bad as an actual code issue, because the consequence for users is the same: The performance is worse than what has been demonstrated in pre-release QA testing.

MaEtUgR, 2019-07-16:
Just wanted to mention the pr for UX parameter default changes here because it seems related: https://github.com/PX4/Firmware/pull/12346

jkflying, 2019-07-18:
Another potential one is `THR_MDL_FAC`, it defaults to 0 which is a linear relationship between PWM and thrust. A better value would be ~0.65, to take into account some linearities, and make the PIDs actually work at different thrust values.

julianoes, 2019-07-24:
Notes from devcall:

- @RomanBapst: there might be one more param for trigger interval
- @RomanBapst: will be done when the PR is merged, presumably it will go into 1.10 since 1.10 should be released in 5 weeks already.

PR #12543 (2019-07-29): Update default parameters
See  #12361  for context.

Also see individual commit message for some explanations.

fixes #12361
closed 2019-07-29 by dagar