PX4/PX4-Autopilot #12362: PID implementation - Parallel vs Ideal

bresch, 2019-06-28:
I wanted to discuss a bit in this issue the different types on PID controllers. The main reason is that I find the current implementation not intuitive to adjust manually.

**Parallel:**
This is the current implementation of the rate and velocity controllers in PX4. The specificity of that implementation is that the P, I and D path are independent. This sounds to be a good idea, but in practice it is not practical because modifying the proportional gain (P) changes the integral and derivatives actions. Consequence: both decreasing or increasing the P gain can destabilize the system because decreasing P increases the integral effect.
[image]

**Ideal:**
This implementation is widely used in commercial controller as it decouples the integral and derivative actions from the proportional gain: The gain of the controller is defined by the P gain, the integral action by its time constant Ti (or Ki' = 1/Ti) and the derivative action by Td (or Kd' = Td).
[image]

**Describe problem solved by the proposed feature**
Adjusting a parallel PID controller requires to modify all three gains together. This a a reason why the tuning sequence is "Set I and D to zero, tune P and D together and then add a bit of I to reduce the steady-state error." and that some vehicles are poorly tuned because "the integrator is scary".

Changing to the ideal form should make the tuning easier. In order to show why it is not intuitive, I collected some rate controller gains of the standard config files, computed the integral and derivative effects and plotted the integral effect with the Ki gain (scaled up) on the same graph.
On the graph, you can directly see that more Ki doesn't necessarily mean "more integral action". 

Take for example the M100 and the Mid size quad 1: the M100 has 20% more integral action but it's Ki gain is three times smaller. Also, its derivative time constant is only 50% smaller even if Kd is 7 times smaller. This is simply because the proportional gain of those two vehicles are quite different, mainly due to different propulsion systems and inertia. Furthermore, Td is quite constant across the frames but Kd have to be tuned every time Kp changes.

I am convinced that a simple slider that changes the Kp' gain of an ideal PID would be enough to tune most of the vehicles.

[image]
[image]

**Implementation**
We can easily convert the actual algorithm into a Ideal form by introducing a "Controller gain" Kc that multiplies Kp, Ki and Kd. This Kc can then be mapped to a slider in QGC to increase or decrease the performance/robustness of the controller.

FYI: @bkueng

LorenzMeier, 2019-06-28:
Very cool! Let's make sure we can execute on this on a fixed timeline.

jlecoeur, 2019-06-29:
Sounds very cool!

I was curious and displayed the gains you collected:
[image]

For the derivative part (left), it indeed looks like Kd was tuned together with Kp. Sounds like Kc=Kp, Kp'=1, Kd'=0.029 would be a good starting point.

For the integrator, however, there is no such correlation between Kp and Ki, two options:
- the platforms were not properly tuned (as you said "the integrator is scarry")
- the platforms were properly tuned, and it is better not to scale the integral action with Kc

LorenzMeier, 2019-06-29:
@jlecoeur It is important to note that we have protected against large I gains with this commit, so tuning can now include a large I gain:
https://github.com/PX4/Firmware/commit/ea31f34d09b542fcad7087aa9c9d9c27ddad6ce9

I would also argue that a lot of airframe configs here would not stand an evaluation on what we consider "great performance", as we find that community-provided tuning values are hit-and-miss. My take-away is that we need to centralize these configs and make sure we have the same scrutiny on them as we have on code changes.

bresch, 2019-07-01:
@jlecoeur Thanks for your input and the graphs. Yes, there is a quite clear linear relation between Kp and Kd.
There is no visible linear relation between Kp and Ki because Kp mainly depends on total thrust and inertia ratio while Ki depends essentially on the inertia of the vehicle. 
- Weak and heavy vehicle: high Kp, low integral
- Powerful and heavy vehicle: low Kp, low integral
- Weak and light vehicle: high Kp, high integral
- Powerful and light vehicle, high Kp and high integral

(when I say "low integral", understand low 1/T1 or low Ki')

In summary, the integral time constant is more correlated with the size of the vehicle than Kp. This is why in the table of my original post (sorted by increasing 1/T1), the VTOL plane is at the top, the medium size drones at the middle and the racers at the bottom while the Kp gains seem more random (even if a racer usually have a much higher thrust-to-weight ratio than other drones).

I'll add a new variable for the "controller gain" with a default of 1.0 and try with different platforms to see if it's really easier to tune in practice. Since it only scales the other gains, there will be no visible changes for the current setups.

jlecoeur, 2019-07-01:
@bresch sounds good!

If it turns out that you often need to tune Kc and Ki separately then you may consider an "ideal-parallel" implementation where Kc scales both the proportional and derivactive actions, but not the integral action.

I

bresch, 2019-07-01:
@jlecoeur In fact you still want to Ki with Kc because this decouples the static part (pure gain) from the dynamic tuning (integral and derivative time constants).

**Intuitive example**
Let's look at the following scenario:
You have a drone well tuned for you application with the gains Kp, Ki and Kd. Now, you want to update your lifting system to use 6S batteries instead of 4S and even if you changed the motors with a lower KV, the gain of the system is now higher. To obtain the same behavior as before, you have to adjust the gains. 

- Parallel PID: In parallel form, you would have to adjust together Kp, Ki and Kd (reducing Kp only might create oscillations). You will mot likely en up setting Ki to zero and start the tuning from scratch.

- Ideal PID: Since the Kp' gain controls the overall gain of the controller, the only adjustment you need to do is to reduce this gain in order to keep the product `Kp' x drone gain` constant. Since nothing else changed on the drone there is no reason why the integral and derivatives time constants should change.

**Proof**
A more theoretical proof can be shown by inspecting the transfer functions of both Ideal and Parallel PID controllers:
<a href="https://www.codecogs.com/eqnedit.php?latex=\\&space;C_{ideal}(s)=K_p'K_i'K_d'\frac{(s&plus;K_i')(s&plus;1/K_d')}{s}&space;\\&space;C_{parallel}(s)=\frac{K_iK_d}{K_p}\frac{(s&plus;K_i/K_p)(s&plus;K_p/K_d)}{s}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\\&space;C_{ideal}(s)=K_p'K_i'K_d'\frac{(s&plus;K_i')(s&plus;1/K_d')}{s}&space;\\&space;C_{parallel}(s)=\frac{K_iK_d}{K_p}\frac{(s&plus;K_i/K_p)(s&plus;K_p/K_d)}{s}" title="\\ C_{ideal}(s)=K_p'K_i'K_d'\frac{(s+K_i')(s+1/K_d')}{s} \\ C_{parallel}(s)=\frac{K_iK_d}{K_p}\frac{(s+K_i/K_p)(s+K_p/K_d)}{s}" /></a>
We clearly see that the zeros of the ideal transfer function are defined by Ki' and Kd' while in the parallel form, the proportional gain affects the position of the zeros.

jlecoeur, 2019-07-01:
Very clear explanation, thank you! Looking forward to seeing the PR ;-)

PR #12472 (2019-07-16): PID rate controller - Add controller gain to support Ideal PID form (ISA standard)
The introduction of a parameter that controls the total gain of the controller facilitate the tuning of a new airframe as it decouples properly the integral and derivative time constants. A user should be able to take the gains of a similar platform (size/inertia) and simply adjust this gain to have it flying properly. 
Until now, with the pure parallel form, if the thrust to weight ratio is different from the reference config, the tuning has to be done almost from scratch.

The default value is 1.0 and basicall [... cut]
closed 2019-07-16 by bresch