PX4/PX4-Autopilot #12375: New fw 1.9.1 doesn't recognize airspeed sensor

DamDrone, 2019-06-30:
Hi guys,
I have a twin motor fixed wing drone equipped with Pixhawk 2.4.6 by mRobotics.
I flew this drone until yesterday succesfully with the fw 1.8, today I flashed to fw 1.9.1 and suddendly, without any change on the parameters, the system doesn't recognize the airspeed sensor anymore.
The airspeed sensor is this one:
https://store.mrobotics.io/mRo-Next-Gen-MS5525-Airspeed-Sensor-NEW-p/mro-ms5525v2-mr.htm

It is connected to i2c bus on a splitter, on the same splitter there are:
- Pixhawk 2.4.6 by mrobotics
- compass of this GPS module: https://store.mrobotics.io/mRo-GPS-u-Blox-Neo-M8N-HMC5983-Compass-p/mro-gps004-mr.htm
- Rangefinder: Garmin LidarLite V3 (already setted up on 1.8 and working)

What could be the issue that makes the airspeed not visible?

Thanks for an help

dagar, 2019-06-30:
You're likely running the px4_fmu-v2 (1MB flash limit) binary rather than px4_fmu-v3 (full 2MB flash).

Here's the board config for the flash limited px4_fmu-v2 where various drivers and modules are not included to save space. https://github.com/PX4/Firmware/blob/d1298f2c372147eb6142c102c9499a2be6b96350/boards/px4/fmu-v2/default.cmake#L29

One way to verify is to run `ver all` in a console (QGC mavlink shell) or http://dev.px4.io/en/debug/system_console.html. You'll likely be able to upload your bootloader and then start flashing the full featured px4_fmu-v3 binary automatically.

dagar, 2019-06-30:
@pkocmoud can you verify the bootloader version you ship with?

DamDrone, 2019-06-30:
Thanks for the answer, sadly I have zero knowledge about what goes more than flash a firmware or modify the parameters through QGC, so I have no idea how to use links and files you just posted.

Is there any kind of file that I can flash like custom firmware flashing, that include already the mods you posted?

Look like my pixhawk has v3 fmu, if I checked correctly on the web page where I bought it:
https://store.mrobotics.io/Genuine-PixHawk-Flight-Controller-p/mro-pixhawk1-minkit-mr.htm

DamDrone, 2019-06-30:
Here is what appear on setting the command you said
[image]

dagar, 2019-06-30:
Ok, so you are running px4_fmu-v3 with the proper bootloader (false alarm @pkocmoud). Something else must be wrong with either the ms5525 driver or the surrounding startup code. I'll try to reproduce locally.

DamDrone, 2019-06-30:
Thank you, I just re-flashed with the fw for fmuv3 but still getting the issue.
Will wait for your feedback, thanks!

DamDrone, 2019-07-02:
@dagar Hi, any news about?

DamDrone, 2019-07-07:
@dagar Hi Daniel, something going on? I'm grounded and eventually need to downgrade...

mcsauder, 2019-07-07:
Hi @DamDrone , can you report back what the value of your autopilot's `CBRK_AIRSPD_CHK` parameter is?

dagar, 2019-07-09:
@DamDrone in the console can you try starting it manually?

`ms5525 start`
`listener differential_pressure`

mcsauder, 2019-07-09:
Hi @DamDrone , find me on PX4 slack so I can help you out.

DamDrone, 2019-07-10:
@mcsauder CBRK_AIRSPD_CHK is set to 0
@dagar it say “command not found”

mcsauder, 2019-07-10:
@dagar and @DamDrone , I dusted off my pixhawk (fmu-v2), loaded v1.8.1 firmware, set the airframe type to standard fixed wing, and verified that the ms4525 sensor I have started automatically, which it did.  I then upgraded to v1.9.2 via QGC and re-validated.  (I did need to calibrate my airspeed sensor, but the driver started appropriately.)

@DamDrone , I do not have an fmu-v3 or an ms5525 to test with, but I think perhaps this might just be a configuration and calibration issue.  I know when we talked on slack you were already at your time limit for getting back to flight status after the firmware upgrade, but in case you want to try to make it work this is what I would suggest:

1) Upgrade your firmware via QGC again so that you have v1.9.2
2) **Make sure** that your airspeed sensor is physically connected to the autopilot (just in case).
3) Follow the airframe configuration instructions and ensure you have a standard fixed wing airframe selected: https://docs.px4.io/en/config/airframe.html
4) Go to the sensors tab in QGC, you should have an option to calibrate your airspeed sensor. If not, it is still not seeing your sensor on the bus and/or the driver didn't start.  If it is there, It is probably also a good time to recalibrate all of your other sensors as well. https://docs.px4.io/en/config/airspeed.html

Following our conversation in slack, you can give this issue a few more days and close it out once you are done troubleshooting.

(@dagar, listener must be disabled in fmu-v2, it doesn't work for me either.)

mcsauder, 2019-07-20:
@DamDrone , I ordered an ms5525 and was able to test with it today.  The driver starts fine with fmu-v5 hardware, but does not with fmu-v2.
[image]

This is because the default version of firmware installed by QGroundControl is not compiling the ms5525 driver into the binary file in v1.9.2.  

It is getting compiled into the px4_fmu-v2_fixedwing binary.

[image]

This is because of the flash size limitation for your hardware.

You'll need to load the px4_fmu-v2_fixedwing binary onto your autopilot for your sensor to work.

closed 2019-07-20 by mcsauder