PX4/PX4-Autopilot #12383: Px4(1.9.0) support the offboard mode for fixed wing or not?

zhanghouxin07, 2019-07-01:
Hi there,

I'am research the fixed wing platform about PX4. However, I can enter the offboard mode with FW in simulation. The Developer guide says that Fw is not supported by PX4, but someone say the FW can be used on FW.

Thank you for your help!!

Thanks!

hamishwillee, 2019-07-02:
@RomanBapst Can you confirm whether this is correct, and if not, in which version offboard control for fixed wing was implemented. In https://docs.px4.io/master/en/flight_modes/offboard.html we say:
[image]

zhanghouxin07, 2019-07-02:
I try to enter the offboard mode again. I complete it successfully. the developer guide of px4 may be a little out of style. 
thx

closed 2019-07-02 by zhanghouxin07

Jaeyoung-Lim, 2019-07-02:
@hamishwillee Fixed wing offboard mode has only started work since #12149 and #12311 

Only attitude setpoints (attitude, bodyrates) are supported at this point.

hamishwillee, 2019-07-03:
@Jaeyoung-Lim Thanks very much - I've added docs in https://github.com/PX4/px4_user_guide/pull/506

As a general rule, an undocumented feature is no feature at all :-). So if you add something, please document, ask me for help, or as a last resort add a docs issue.

Jaeyoung-Lim, 2019-07-03:
@hamishwillee Thanks for the input. I will try to make more of the features I work on to be shown in the documentation