PX4/PX4-Autopilot #12385: 'px4_sitl gazebo' errors when performing third build step

dyli96, 2019-07-01:
Hi there,
I ran ```make px4_sitl gazebo``` and got tons of errors in the third build step.
I ran ```source ubuntu_sim.sh``` before making
I also followed the dependencies for ROS/Gazebo.
I can run ```make px4_sitl_default jmavsim``` correctly but got the errors for gazebo.
Thanks for your help!
```
[3/6] Performing build step for 'sitl_gazebo'
FAILED: cd /home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/csgurel/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/50] Building CXX object CMakeFiles/sensor_msgs.dir/Imu.pb.cc.o
[2/50] Linking CXX shared library libmav_msgs.so
[3/50] Linking CXX shared library libstd_msgs.so
[4/50] Building CXX object CMakeFiles/sensor_msgs.dir/Range.pb.cc.o
[5/50] Building CXX object CMakeFiles/sensor_msgs.dir/SITLGps.pb.cc.o
[6/50] Building CXX object CMakeFiles/sensor_msgs.dir/OpticalFlow.pb.cc.o
[7/50] Building CXX object CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o
[8/50] Linking CXX static library OpticalFlow/klt_feature_tracker/libklt_feature_tracker.a
[9/50] Building CXX object CMakeFiles/physics_msgs.dir/Wind.pb.cc.o
[10/50] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/optical_flow.cpp.o
[11/50] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/px4flow.cpp.o
[12/50] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_px4.cpp.o
[13/50] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o
FAILED: /usr/bin/c++   -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -Dphysics_msgs_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/csgurel/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/csgurel/src/Firmware/mavlink/include -I/usr/include/Paging -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv -I/home/csgurel/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++14 -MMD -MT CMakeFiles/physics_msgs.dir/Wind.pb.cc.o -MF CMakeFiles/physics_msgs.dir/Wind.pb.cc.o.d -o CMakeFiles/physics_msgs.dir/Wind.pb.cc.o -c Wind.pb.cc
In file included from Wind.pb.h:32:0,
                 from Wind.pb.cc:5:
/usr/include/gazebo-9/gazebo/msgs/vector3d.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-9/gazebo/msgs/vector3d.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-9/gazebo/msgs/vector3d.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from Wind.pb.cc:5:0:
Wind.pb.h: In member function ‘const gazebo::msgs::Vector3d& physics_msgs::msgs::Wind::force() const’:
Wind.pb.h:318:8: error: ‘_Vector3d_default_instance_’ is not a member of ‘gazebo::msgs’
       &::gazebo::msgs::_Vector3d_default_instance_);
        ^
Wind.pb.cc: In static member function ‘static void physics_msgs::msgs::protobuf_Wind_2eproto::TableStruct::InitDefaultsImpl()’:
Wind.pb.cc:98:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::InitDefaults();
                   ^
Wind.pb.cc:102:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
Wind.pb.cc: In function ‘void physics_msgs::msgs::protobuf_Wind_2eproto::{anonymous}::AddDescriptorsImpl()’:
Wind.pb.cc:122:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::AddDescriptors();
                   ^
FAILED: /usr/bin/c++   -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -Dsensor_msgs_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/csgurel/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/csgurel/src/Firmware/mavlink/include -I/usr/include/Paging -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv -I/home/csgurel/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++14 -MMD -MT CMakeFiles/sensor_msgs.dir/Imu.pb.cc.o -MF CMakeFiles/sensor_msgs.dir/Imu.pb.cc.o.d -o CMakeFiles/sensor_msgs.dir/Imu.pb.cc.o -c Imu.pb.cc
In file included from Imu.pb.h:32:0,
                 from Imu.pb.cc:5:
/usr/include/gazebo-9/gazebo/msgs/quaternion.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-9/gazebo/msgs/quaternion.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-9/gazebo/msgs/quaternion.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from Imu.pb.h:33:0,
                 from Imu.pb.cc:5:
/usr/include/gazebo-9/gazebo/msgs/vector3d.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/gazebo-9/gazebo/msgs/vector3d.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
[... 85 lines cut ...]
                   ^
Odometry.pb.cc:111:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
Odometry.pb.cc:113:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Quaternion’
       ::gazebo::msgs::Quaternion::internal_default_instance());
       ^
Odometry.pb.cc:115:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
Odometry.pb.cc:117:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
Odometry.pb.cc: In function ‘void nav_msgs::msgs::protobuf_Odometry_2eproto::{anonymous}::AddDescriptorsImpl()’:
Odometry.pb.cc:142:19: error: ‘gazebo::msgs::protobuf_quaternion_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_quaternion_2eproto::AddDescriptors();
                   ^
Odometry.pb.cc:143:19: error: ‘gazebo::msgs::protobuf_vector3d_2eproto’ has not been declared
   ::gazebo::msgs::protobuf_vector3d_2eproto::AddDescriptors();
                   ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:197: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```

julianoes, 2019-07-02:
Have you tried `make clean` and then building it again?
And can you please post the complete build output?

dyli96, 2019-07-02:
I updated my protoc and now have new errors.
The errors are too long. I attached a file.

[px4_sitl_gazebo_errors.txt](https://github.com/PX4/Firmware/files/3352117/px4_sitl_gazebo_errors.txt)

julianoes, 2019-07-02:
That's odd. Have you checked if your system packages are all up to date e.g. `apt-get update && apt-get upgrade`?

On which Ubuntu (or other distribution) are you?

dyli96, 2019-07-03:
I tried ```apt-get update && apt-get upgrade```, it did upgrade something. After ```make clean```, I still got the same errors.
It's Ubuntu 18.04.2 LTS

julianoes, 2019-07-04:
Have you added any custom ppa's for things like `protoc` or `gazebo`?

julianoes, 2019-10-02:
Closing because there was no follow up.

pranav083, 2019-10-07:
i am having the same version conflict problem of "protoc" while running the command :
# Start Gazebo with plane
make px4_sitl gazebo_plane
I don't know what to do. Any suggestion will be helpful.

hamishwillee, 2019-10-07:
@julianoes Reopened. @pranav083 is your setup EXACTLY as described above? If not, how does it differ. What does your your build log look like. Can you respond to https://github.com/PX4/Firmware/issues/12385#issuecomment-508343494

julianoes, 2019-10-08:
@pranav083 are you also on Ubuntu 18.04?

pranav083, 2019-10-09:
sorry, for the late reply ,But yes i on ubuntu 18.04 .And this is the error that i am getting 
https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1

pranav083, 2019-10-09:
> @julianoes Reopened. @pranav083 is your setup EXACTLY as described above? If not, how does it differ. What does your your build log look like. Can you respond to [#12385 (comment)](https://github.com/PX4/Firmware/issues/12385#issuecomment-508343494)

yes , @hamishwillee i had run the same command as it was stated above.

julianoes, 2019-10-09:
@pranav083 can you try to do these steps and paste the output here:
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libprotobuf-dev
```

pranav083, 2019-10-09:
@julianoes I had post the updated error in the same link:
https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1#file-gazebo-make-error-updated

julianoes, 2019-10-09:
You need to figure out the protobuf packages installed and where they come from somehow. I have seen this problem before but unfortunately we don't remember how we fixed it.

What's the output of `apt-cache showpkg libprotobuf-dev`?

@kamilritz do you remember?

pranav083, 2019-10-09:
here is the output of the command `apt-cache showpkg `libprotobuf-dev` https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1#file-showpkg-libprotobuf-dev

kamilritz, 2019-10-09:
I also can not recall exactly and I do not have my computer with me at the moment. 
Did you run `make clean` after installing `libprotobuf-dev`. I think you need version 3.0.0. But during the first run your console output shows:
``` 
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread;-lpthread;-lpthread (found version "3.4.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.4.1
```
I guess you will therefore need to run `make clean` first.

pranav083, 2019-10-09:
This time  , I had run all the command step by step after restarting my laptop and make a fresh copy of the clone of Firmware from github . https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1#file-updated-error

It is still returning the same error . But , one thing that i had noticed is that during the execution of the make file with the command `make px4_sitl gazebo`
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread

In the line no. 116- 117 of the above link . I had searched for the error but doesn't able  to understand its significance .

kamilritz, 2019-10-09:
It is still finding the wrong version of libprotobuf
```
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread;-lpthread;-lpthread (found version "3.4.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.4.1
```
I am pretty sure, that this is the issue. It needs to find the protobuf version 3.0.0 which you install with the apt-get command.

You could try to remove the `/usr/local/lib/libprotobuf.a`file.

kamilritz, 2019-10-09:
Now it found `/usr/lib/x86_64-linux-gnu/libprotobuf.so`. You may want to remove also this. I remember to have removed a few files, but no guarantees.

pranav083, 2019-10-09:
Now , Its is return this 
`-- Found Protobuf: Protobuf_LIBRARY-NOTFOUND;-lpthread;-lpthread;-lpthread (found version "3.4.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.4.1
`
and for more error see:
https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1#file-updated-error

kamilritz, 2019-10-09:
Ok, that seems to be a dead end. I will check on my computer and come back to you tomorrow.

pranav083, 2019-10-09:
ok.

> Ok, that seems to be a dead end. I will check on my computer and come back to you tomorrow.

kamilritz, 2019-10-10:
Can you try to look which protobuf related files are on  your computer. Try `sudo find / -type f -name 'libproto*'`

My output looks roughly like this:
```
/var/lib/dpkg/info/libprotobuf-lite19:amd64.list
/var/lib/dpkg/info/libprotoc19:amd64.list
/var/lib/dpkg/info/libprotoc-dev:amd64.list
/var/lib/dpkg/info/libprotobuf-lite10:amd64.list
/var/lib/dpkg/info/libprotobuf-dev:amd64.list
/var/lib/dpkg/info/libprotoc10:amd64.list
/var/lib/dpkg/info/libprotobuf10:amd64.list
/var/lib/dpkg/info/libprotobuf10:amd64.md5sums
/var/lib/dpkg/info/libprotobuf10:amd64.shlibs
/var/lib/dpkg/info/libprotobuf10:amd64.triggers
/var/lib/dpkg/info/libprotobuf19:amd64.list
/var/lib/dpkg/info/libprotobuf19:amd64.md5sums
/var/lib/dpkg/info/libprotobuf19:amd64.shlibs
/var/lib/dpkg/info/libprotobuf19:amd64.triggers
/var/lib/dpkg/info/libprotobuf-lite19:amd64.md5sums
/var/lib/dpkg/info/libprotobuf-lite19:amd64.shlibs
/var/lib/dpkg/info/libprotobuf-lite19:amd64.triggers
/var/lib/dpkg/info/libprotoc19:amd64.md5sums
/var/lib/dpkg/info/libprotoc19:amd64.shlibs
/var/lib/dpkg/info/libprotoc19:amd64.triggers
/var/lib/dpkg/info/libprotobuf-lite10:amd64.md5sums
/var/lib/dpkg/info/libprotobuf-lite10:amd64.shlibs
/var/lib/dpkg/info/libprotobuf-lite10:amd64.triggers
/var/lib/dpkg/info/libprotobuf-dev:amd64.md5sums
/var/lib/dpkg/info/libprotoc-dev:amd64.md5sums
/var/lib/dpkg/info/libprotoc10:amd64.md5sums
/var/lib/dpkg/info/libprotoc10:amd64.shlibs
/var/lib/dpkg/info/libprotoc10:amd64.triggers
/usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.19.0.0
/usr/lib/x86_64-linux-gnu/libprotoc.a
/usr/lib/x86_64-linux-gnu/libprotoc.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotoc.so.19.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf.so.19.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf-lite.a
/usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf.a
/usr/lib/libreoffice/program/libprotocolhandlerlo.so
/usr/lib/pulse-11.1/modules/libprotocol-http.so
/usr/lib/pulse-11.1/modules/libprotocol-cli.so
/usr/lib/pulse-11.1/modules/libprotocol-simple.so
/usr/lib/pulse-11.1/modules/libprotocol-native.so
```
My `libprotobuf-dev` and `libprotoc-dev ` are on version 3.0.0  and while running `make px4_sitl gazebo` it is detecting the version 3.0.0. correctly.

pranav083, 2019-10-10:
yes , this is my output of the search result. `sudo find / -type f -name 'libproto*'` is 
`mighty@ideapad:~$ sudo find / -type f -name 'libproto*'
[sudo] password for mighty: 
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/cmake/libprotoc.cmake
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/cmake/libprotobuf.cmake
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/cmake/libprotobuf-lite.cmake
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/src/libprotobuf.map
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/src/libprotoc.map
/home/mighty/Downloads/protobuf-all-3.10.0/protobuf-3.10.0/src/libprotobuf-lite.map
/home/mighty/ros_ws/ai_ws/protobuf/cmake/libprotoc.cmake
/home/mighty/ros_ws/ai_ws/protobuf/cmake/libprotobuf.cmake
/home/mighty/ros_ws/ai_ws/protobuf/cmake/libprotobuf-lite.cmake
/home/mighty/ros_ws/ai_ws/protobuf/build/libprotobuf-lite.a
/home/mighty/ros_ws/ai_ws/protobuf/build/libprotobuf.a
/home/mighty/ros_ws/ai_ws/protobuf/build/libprotoc.a
/home/mighty/ros_ws/ai_ws/protobuf/src/libprotobuf.map
/home/mighty/ros_ws/ai_ws/protobuf/src/libprotoc.map
/home/mighty/ros_ws/ai_ws/protobuf/src/libprotobuf-lite.map
/var/lib/dpkg/info/libprotobuf10:amd64.list
/var/lib/dpkg/info/libprotobuf10:amd64.md5sums
/var/lib/dpkg/info/libprotobuf10:amd64.shlibs
/var/lib/dpkg/info/libprotobuf10:amd64.triggers
/var/lib/dpkg/info/libprotoc10:amd64.md5sums
/var/lib/dpkg/info/libprotoc10:amd64.shlibs
/var/lib/dpkg/info/libprotoc10:amd64.triggers
/var/lib/dpkg/info/libprotoc10:amd64.list
/var/lib/dpkg/info/libprotobuf-lite10:amd64.md5sums
/var/lib/dpkg/info/libprotobuf-lite10:amd64.shlibs
/var/lib/dpkg/info/libprotobuf-lite10:amd64.triggers
/var/lib/dpkg/info/libprotobuf-lite10:amd64.list
/var/lib/dpkg/info/libprotobuf-dev:amd64.md5sums
/var/lib/dpkg/info/libprotobuf-dev:amd64.list
/var/lib/dpkg/info/libprotoc-dev:amd64.md5sums
/var/lib/dpkg/info/libprotoc-dev:amd64.list
/snap/gnome-3-28-1804/71/usr/lib/pulse-11.1/modules/libprotocol-cli.so
/snap/gnome-3-28-1804/71/usr/lib/pulse-11.1/modules/libprotocol-http.so
/snap/gnome-3-28-1804/71/usr/lib/pulse-11.1/modules/libprotocol-native.so
/snap/gnome-3-28-1804/71/usr/lib/pulse-11.1/modules/libprotocol-simple.so
/snap/gnome-3-28-1804/67/usr/lib/pulse-11.1/modules/libprotocol-cli.so
/snap/gnome-3-28-1804/67/usr/lib/pulse-11.1/modules/libprotocol-http.so
/snap/gnome-3-28-1804/67/usr/lib/pulse-11.1/modules/libprotocol-native.so
/snap/gnome-3-28-1804/67/usr/lib/pulse-11.1/modules/libprotocol-simple.so
/usr/local/lib/libprotobuf-lite.a
/usr/local/lib/libprotoc.a
/usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf-lite.a
/usr/lib/x86_64-linux-gnu/libprotoc.so.10.0.0
/usr/lib/x86_64-linux-gnu/libprotobuf.a
/usr/lib/x86_64-linux-gnu/libprotoc.a
/usr/lib/pulse-11.1/modules/libprotocol-http.so
/usr/lib/pulse-11.1/modules/libprotocol-simple.so
/usr/lib/pulse-11.1/modules/libprotocol-native.so
/usr/lib/pulse-11.1/modules/libprotocol-cli.so
find: ‘/run/user/1000/gvfs’: Permission denied
`

kamilritz, 2019-10-10:
Can you upload the output such that each file is on a separate line?

julianoes, 2019-10-10:
Don't forget to do `sudo ldconfig` after deleting files.

kamilritz, 2019-10-10:
Thanks. I would start with removing the libraries that are in your ros workspace, of course only if you do not need them for something else. The libraries in `/snap/` should be okay, I have them too.
And then also remove other libraries that do not appear on my list. For sure the ones in `/usr/local/lib/`.
And before trying again do `sudo ldconfig` as julianoes suggested.

pranav083, 2019-10-10:
now , I have run the command as follows see the command that i had gone through, If you find any error and also see line no. `273 to 275`:
https://gist.github.com/pranav083/c005375b0d663a8f917a12224b2aadd1#file-after-implentation
It is getting the same version of the `libprotobuf` . Is their any way to downgrade my libprotobuf to the desired library version.

kamilritz, 2019-10-10:
You still have some libproto* files /home/mighty/. I would suggest to do the standard installation again after removing certain libraries. Because you can not be sure that you did not remove a wrong library file. This includes also to reinstall gazebo. Just do again these parts of the toolchain installation.

pranav083, 2019-10-12:
I had finally solve this error . It was definitely a `libprotobuf`version error as the version that is needed to build is `version 3.0.0` and mine was detecting a upgarded version that was `3.4.1`. 
As stated above one can check his version by running `protoc --version`  on your terminal. The `libprotobuf` is actually installed in the directory `/usr/local/include/google`.
So, to installed the correct version first download the correct file from here  [protobuf-cpp-3.0.0.tar.gz](https://github.com/google/protobuf/releases/download/v3.0.0/protobuf-cpp-3.0.0.tar.gz) . Into your Download folder. And then follow the steps that are listed on this link :  

http://answers.gazebosim.org/question/18096/this-file-was-generated-by-an-older-version-of-protoc/  

**NOTE :** In the above link file you have downloaded their is no need of running `$ ./autogen.sh`  as the required file are already downloaded in it .It is due to outdated `autogen.sh` file as listed [here](https://github.com/grpc/grpc/issues/7952) . 
And also when you run `$ make check` it will return some error . But, you can ignore that as it run fine  without that also.
After that go to Your Firmware directory that you had cloned. And run 
`$ make clean`
`$ make px4_sitl gazebo`
Now, you are good to go.
Hope, this help others

julianoes, 2019-10-14:
Nice, thanks for sharing this. I guess the problem was the version that was somehow installed to `/usr/local`. I would think that you can just remove what is conflicting from `/usr/local` and then install the default one again which comes with the distribution.

closed 2019-10-14 by julianoes

bigbellmercy, 2020-09-10:
> I had finally solve this error . It was definitely a `libprotobuf`version error as the version that is needed to build is `version 3.0.0` and mine was detecting a upgarded version that was `3.4.1`. [...]

For similar users,

### A solution case
In my case `cartographer_ros` caused this [`protoc` version collision problem](https://answers.gazebosim.org//question/18096/this-file-was-generated-by-an-older-version-of-protoc/). So, I did the above method but it failed in my case.

So, as like the another [solution](https://github.com/PX4/Firmware/issues/11568#issuecomment-468769608), I reinstalled [ROS/Gajebo](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo) again and the problem disappeared.

Iko-Git, 2021-01-06:
> > I had finally solve this error . It was definitely a `libprotobuf`version error as the version that is needed to build is `version 3.0.0` and mine was detecting a upgarded version that was `3.4.1`.
> > As stated above one can check his version by running `protoc --version` on your terminal. The `libprotobuf` is actually installed in the directory `/usr/local/include/google`.
> > So, to installed the correct version first download the correct file from here [protobuf-cpp-3.0.0.tar.gz](https://github.com/google/protobuf/releases/download/v3.0.0/protobuf-cpp-3.0.0.tar.gz) . Into your Download folder. And then follow the steps that are listed on this link :
> > http://answers.gazebosim.org/question/18096/this-file-was-generated-by-an-older-version-of-protoc/
> > **NOTE :** In the above link file you have downloaded their is no need of running `$ ./autogen.sh` as the required file are already downloaded in it .It is due to outdated `autogen.sh` file as listed [here](https://github.com/grpc/grpc/issues/7952) .
> > And also when you run `$ make check` it will return some error . But, you can ignore that as it run fine without that also.
> > After that go to Your Firmware directory that you had cloned. And run
> > `$ make clean`
> > `$ make px4_sitl gazebo`
> > Now, you are good to go.
> > Hope, this help others
> 
> For similar users,
> 
> ### A solution case
> In my case `cartographer_ros` caused this [`protoc` version collision problem](https://answers.gazebosim.org//question/18096/this-file-was-generated-by-an-older-version-of-protoc/). So, I did the above method but it failed in my case.
> 
> So, as like the another solution in #11568, I reinstalled [ROS/Gajebo](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo) again and the problem disappeared.

I am facing the same problem. Could you please guide me in the installation of libprotobuf version 3.0.0?!

bigbellmercy, 2021-01-06:
I'm sorry. I fortgot to install it. May someone help you..

Iko-Git, 2021-01-06:
> I'm sorry. I fortgot to install it. May someone help you..

The idea is I need guidance in general to solve the problem. I spent a lot of hours and am blocking in, in the following link you can find the error I am receiving while running make px4_sitl_default gazebo: https://discuss.px4.io/t/error-while-running-make-px4-sitl-default-gazebo/20388

Iko-Git, 2021-01-06:
> The idea is I need guidance in general to solve the problem. I spent a lot of hours and am blocking in, in the following link you can find the error I am receiving while running make [...]

I applied `pip install protobuf==3.0.0` `sudo ldconfig` `make clean` then still facing the same problem while running `make px4_sitl_default gazebo`

addy1997, 2021-01-26:
@Iko-Git thanks a ton. Your solution worked for me. I ran into this error after installing ignition gazebo on my ubuntu 18.04.2 LTS. 

####CAUTION####

only run the commands given below if you have following system configurations like me:

1. OS: Mac OSX (Big Sur)
2. Running gazebo on **Ubuntu 18.04.2 LTS (bionic beaver) using Virtual Machine.**

**How I got it right?**

1. `cd PX4-Autopilot`
2. `make clean `
3. `make px4_sitl jmavsim`
4. `make px4_sitl gazebo`

**After doing this, my gazebo was still not appearing properly. So to get rid of the glitch and delay, I did this**

1. `export SVGA_VGPU10=0` #following environmental variable presumably shrinking down the graphics instruction set to OpenGL2.

dbenders1, 2022-05-19:
Just a comment: by following the instructions of [this comment](https://github.com/PX4/PX4-Autopilot/issues/12385#issuecomment-541315457), I got it to work with libprotobuf version 3.6.1.2 (can be downloaded [here](https://github.com/protocolbuffers/protobuf/archive/refs/tags/v3.6.1.2.tar.gz)) for [Stable Release v1.12.3](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.3) on Ubuntu 20.04 and ROS Noetic.