PX4/PX4-Autopilot #12386: 1.9.0: Fixed Wing Mission Rejected

EnsignCrasher, 2019-07-02:
Hi everyone,
I just started using PX4 and everything went very well so far (stabilized, rtl, etc just worked out of the box :+1: )
I then started to try "Mission" and just draw a simple mission with 3 waypoints with RTL in the end. But somehow PX4 always comes up with "REJECT AUTO MISSION" and switches to Loiter.

In the log file seems quite OK for me, except in the end it says `ERROR [PWM3901] driver start failed`, don't know if that is somehow connect or how to resolve this?
I'll upload the ulog later...or ist there any other log about why the mission was rejected?

julianoes, 2019-07-02:
Hi @EnsignCrasher, do you have a log file of when this happens?

EnsignCrasher, 2019-07-02:
jep, here: :relaxed:
https://logs.px4.io/plot_app?log=fbecc4bd-229a-4fb0-8ae1-2555c587d363

julianoes, 2019-07-02:
Is it possible that you took off before you had a decent GPS lock and home position initialized? If you do that then later MISSION and RTL modes are rejected. The fact that MISSION is rejected is not really intuitive and should be fixed. In that case it's the same issue as this: https://github.com/PX4/Firmware/issues/12168

EnsignCrasher, 2019-07-02:
I would think so? Although I haven't checked explicitly. Actually I'm not quite sure how I would check the GPS status the best way in general.
Currently I'm using the pixhawk 4 board, which has no UI LED afaik?
Haven't got around installing telemetry or minimosd and fpv neither :smirk:
And I already found out that connecting the autopilot via USB while trying anything mostly results in strange behaviour :smile:
But I could first try set up the OSD if that might help? That's on my todo-list anyways...
ah, and i think i already tried what is described in the issue ( first land, disarm, and then arm ) and it did not help... but I can recheck that again later

EnsignCrasher, 2019-07-02:
back from the field and it actually seemed to be the issue you mentioned :relaxed: 
just waited for about one or two minutes before arming and everything worked as expected, thanks for your help :relaxed: :+1:

closed 2019-07-02 by EnsignCrasher

julianoes, 2019-07-02:
Ok thanks for reporting. This still needs fixing for better user experience of course.

EnsignCrasher, 2019-07-03:
Guess it would already help if the error message would be a bit more describing.
I would think that there are several checks done, before it fails with "REJECT AUTO MISSION".
probably something like this:
```
execute_mission(){
  if (!check1()){
    goto error;
  }

  if (!check2()){
    goto error;
  }

  run_mission();

  return 0;

error:
  error_message="REJECT AUTO MISSION"
  return 1;
}
```
It would already help a lot, if a description would be added at these places, f.e.
`REJECT AUTO MISSION: no home position was set`
f.e.:
```
 if (!check2()){
    additional_descr="check2 failed, because [...]"
    goto error;
  }

  ....

 error:
    error_message="REJECT AUTO MISSION:" + additional_descr;
    return 1;
```

just an idea .... :relaxed:

julianoes, 2019-07-03:
Yep that's true!

EnsignCrasher, 2019-07-03:
hehe, yeah when I find time in between ;) haven't looked into the code yet though