PX4/PX4-Autopilot #12387: px4::1.9.0 fixed wing in offboard mode but any ros::publish for controling the fw is invalid.

zhanghouxin07, 2019-07-02:
my fixed wing (1.9.0) can enter the offboard mode successfully ,but any publisher by mavros (setpoint_position/local, setpoint_ray/local, or setpoint_attitude ) is invalid.  I cannot control my fw in offboard. 
after entering the offboard mode, the fw do a forward flight only.

i wonder if there are some step I forget to do after entering offborad, but before controlling my fw.

Or, for fw controling, which ros::topic I should publish to?

Thanks!

Jaeyoung-Lim, 2019-07-02:
@zhanghouxin07 I believe the current state is you can send `~setpoint_attitude` setpoints such as attitude, bodyrates. Posiiton setpoints are not valid.

zhanghouxin07, 2019-07-02:
> @zhanghouxin07 I believe the current state is you can send `~setpoint_attitude` setpoints such as attitude, bodyrates. Posiiton setpoints are not valid.

yeah, I note that there are three subtopics in setpoint_attitude:: thrust, cmd_vel, and attitude. Are these three parts used separately or in combination? 
for instance, I send PoseStamped msg to topic::setpoint_attitude/attitude alone. 
e.g: pose.position.x = 1;pose.position.y =0; pose.position.z = 0;
and pose.orientation.x,y,z,w also given for a constant value at the same time. 
but the fw still fly forward.

which steps may be wrong? 
the program tested I have test in rotatedwing platform, it perform successfully. the wrong of code can be excluded.

for more details, the fw first takeoff in misson mode ,then transform in offboard mode. everything is ok but the control of fw. 

thanks for your help, professor!

Jaeyoung-Lim, 2019-07-02:
As I explained, position values are not consumed from the fixed wing position controller. Only the thrust and attitude values will be used. Have you tried using the bodyrate setpoints? Which firmware version are you using? Have you tried with the current master?

zhanghouxin07, 2019-07-02:
thanks for you help again. I have tried using the bodyrate setpoints, and the firmware verison is 1.9.0. I try to achieve goals but failed. 

Here is my main part(not the whole) of my code, I appreciate that you can point out my problem.

      ros::Publisher set_raw_target_atti_pub = nh.advertise<mavros_msgs::AttitudeTarget>("/uav0/mavros/setpoint_raw/target_attitude", 10);
      mavros_msgs::AttitudeTarget  set_raw_target_atti;
//      set_raw_target_atti.type_mask = mavros_msgs::AttitudeTarget::IGNORE_ROLL_RATE | mavros_msgs::AttitudeTarget::IGNORE_PITCH_RATE
//              | mavros_msgs::AttitudeTarget::IGNORE_YAW_RATE;
//      set_raw_target_atti.type_mask = mavros_msgs::AttitudeTarget::IGNORE_ATTITUDE ;
while (ros::ok()){
if (current_state.mode != "OFFBOARD" && (ros::Time::now() - last_request > ros::Duration(5.0)))
                    {
                      if (set_mode_client.call(offb_set_mode) && offb_set_mode.response.mode_sent)
                      {
                        ROS_INFO("Offboard enabled");
                      }
                      last_request = ros::Time::now();
                    }
                    else
                    {
                      if (!current_state.armed && (ros::Time::now() - last_request > ros::Duration(5.0)))
                      {
                        if (arming_client.call(arm_cmd) && arm_cmd.response.success)
                        {
                          ROS_INFO("Vehicle armed");
                        }
                        last_request = ros::Time::now();
                      }
                    }
                quat.setRPY(0,0,180/3.1415926);
                geometry_msgs::Quaternion msg_q;
                tf::quaternionTFToMsg(quat,msg_q);
                set_raw_target_atti.body_rate.x = 0;
                set_raw_target_atti.body_rate.y = 0;
                set_raw_target_atti.body_rate.z = -180/3.1415926;
                set_raw_target_atti.orientation.x = msg_q.x;
                set_raw_target_atti.orientation.y = msg_q.y;
                set_raw_target_atti.orientation.z = msg_q.z;
                set_raw_target_atti.orientation.w = msg_q.w;
                set_raw_target_atti.thrust = 0.8;
                                //pub
                set_raw_target_atti_pub.publish(set_raw_target_atti);
//without the next line, fw cannot enter offboard mode.
                local_pos_pub.publish(pose);
        ros::spinOnce();
}

Jaeyoung-Lim, 2019-07-02:
@zhanghouxin07 It looks like you are missing some fields in the set_raw_target_atti message such as type mask, time stamp

zhanghouxin07, 2019-07-03:
yeah, My program seems to be missing something or some other unrelated codes affecting fw control. I cannot figure out where is the problem. 
Therefore, I rewrite a code for fixed wing. Then, everything is ok with a new code :-). 

thanks a lot.

Jaeyoung-Lim, 2019-07-03:
@zhanghouxin07 Glad to hear that your problem is solved

closed 2019-07-03 by julianoes

hana9090, 2019-08-20:
@Jaeyoung-Lim @zhanghouxin07 
It worked for me to send the local position to vtol in off-board mode! 
using this topic:

`    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);`

But it behaves as multi-rotor not as fixed wing while moving.

Jaeyoung-Lim, 2019-08-20:
@hana9090 I suspect that is because you were sending position setpoints while in multirotor mode

hana9090, 2019-08-20:
How to transit this to fixed wing mode? Okay I will create a new issue. Thank you.

hana9090, 2019-08-20:
@Jaeyoung-Lim here we go
https://github.com/PX4/Firmware/issues/12748

hana9090, 2019-08-20:
@Jaeyoung-Lim  what is the difference here? here send position in a different way but still using off-board mode, right? and the vtol run in fixed-wing mode.

Jaeyoung-Lim, 2019-08-20:
@hana9090 Running vtol in fixed wing mode is not the same as running fixed wing.

Moreover, fixedwing offboard support has been added recently such as https://github.com/PX4/Firmware/pull/12311

zhanghouxin07, 2019-08-20:
> @Jaeyoung-Lim what is the difference here? here send position in a different way but still using off-board mode, right? and the vtol run in fixed-wing mode.

I think maybe you can switch your VTOL from rotate-wing mode to fixed-wing mode first, then make your UAV enter offboard mode. 

I don't know if this is valid. Maybe you can try it.

zhanghouxin07, 2019-08-20:
> @Jaeyoung-Lim what is the difference here? here send position in a different way but still using off-board mode, right? and the vtol run in fixed-wing mode.

I think maybe you can switch your VTOL from rotate-wing mode to fixed-wing mode first by RC, then make your UAV enter offboard mode.

I don't know if this is valid. Maybe you can try it.

Jaeyoung-Lim, 2019-08-20:
@zhanghouxin07 Please test out https://github.com/PX4/Firmware/pull/12532 Position setpoints for fixed wing in offboard mode is being tested