PX4/PX4-Autopilot #12390: sitl_gazebo: throw error if so file for plugin is missing.

lkm1321, 2019-07-02:
Hi there, 

Problem: 
Gazebo doesn't seem throw a noticeable error when it's trying to load a plug-in that simply doesn't exist or isn't built. 
For example, I just spent almost a day or two debugging an issue where Gazebo isn't connecting to the SITL correctly, and it turned out that I wasn't familiar with the changed build configs and didn't build the gazebo interfaces (i.e. `make posix_sitl_default` instead of `make posix_sitl_default gazebo`). As such, the so files that are specified in the URDF didn't exist to start off with.. I suppose this could easily happen with misconfiguring LD_LIBRARY_PATH as well. 

It'll be much nicer to debug if there's an error thrown when the so file is missing. 

I think the nicest solution is if there's something similar to the 'required' tag as if in a roslaunch file. I am not very familiar with URDF files. 

Alternatively, there could be a simple bash/python script that wraps around, e.g. xacro command for generating URDF and spawn_model that only gets 'built' if other gazebo interfaces are built. This way, you won't be able to spawn the vehicle unless the gazebo interface has actually been b uilt.

julianoes, 2019-07-03:
I agree but I think the only way is to look at the logfiles which are mentioned in the log output.

Jaeyoung-Lim, 2021-01-17:
@lkm1321 This is quite obvious now if you use `VERBOSE_SIM=1 ` when starting the SITL simulation with make targets

closed 2021-01-17 by Jaeyoung-Lim