PX4/PX4-Autopilot #12392: Reboot via QGC does not restore dropped RC connection #8148

love-mj, 2019-07-03:


julianoes, 2019-07-03:
It's not clear to me what this is about.

love-mj, 2019-07-03:
Quite often I have the RC control stop working after powering on the autopilot. It doesn't lose signal or indicate lost connection in QGC, it just stops responding to inputs from the RC radio. This only happens before arming, never after. It seems to not happen if I constantly move the sticks pretty regularly, but if left alone for a minute.

I'm not sure why this happens, but at the least if I could reboot via QGC and restore control that would be fine. That doesn't work however, so I have to open a hatch and physically unplug the power. Is there a known reason why the autopilot stops responding to input? If it's a big process to fix the issue, is there a way to restore it via software reboot for the short term?

julianoes, 2019-07-03:
Interesting, we would need to find out what causes this. Can you please provide Firmware version, type of receiver, Pixhawk hardware type, how it is connected. And can you check that the RC receiver is actually still changing the value, e.g. using a scope.

Also, please set https://dev.px4.io/en/advanced/parameter_reference.html#SDLOG_MODE to 1 and then record a log where it happens, upload it to logs.px4.io and then link it here. Thanks!

dagar, 2019-07-03:
Which hardware?

love-mj, 2019-07-03:
all

love-mj, 2019-07-03:
Pixhawk hardware type : pixhawk 2.4.8

love-mj, 2019-07-04:
An electrical reset can be done by either removing power or pushing the FMU reset button the side of Pixhawk next to the USB connector.
You can accomplish a firmware reset by using the CLI to reset the Pixhawk to factor parameters. You should save your existing parameters if you do not want them erased during a firmware reset.

QGC isn't in control of any of this.

davids5, 2019-07-24:
@love-mj  

I have tested this on FMUv5 (Pixhawk 4)
```
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: ea48cd4970977f4d189c1a9ba82483ac03852299
FW version: Release 1.9.0 (17367295)
OS: NuttX
OS version: Release 7.28.0 (119275775)
OS git-hash: 423371c7d4012e725ac4ca51323a18df64e581b3
Build datetime: Jul 24 2019 09:20:02
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000200000000363539373137510100340032
MCU: STM32F76xxx, rev. Z
```

with a Frsky XRS in sbus mode connected to DSM/SBUS RC connector. 

I have tried it with USB alone, and USB and a bench supply connected. 
I have only the Pixhawk4 GPS connected (no lock)

I issues reboot via Mavlink Console and reset via the FMU HW reset button

But my RC continue to work (Tested on the radio cal screen)

Please provide your parameters and a list of what else you have connected. 
If you can also provide the minimum steps to reproduce that will help!

Thank you!

FYI @julianoes , @dagar

julianoes, 2019-07-31:
@love-mj when this happens, could you get the output of `px4io status`?

love-mj, 2019-07-31:
Because I used the qgroundcontrol test, but the results will still appear.You can see https://discussion.px4.io/t/reboot-pixhawk-from-qgc/3508.

julianoes, 2019-07-31:
> https://discussion.px4.io/t/reboot-pixhawk-from-qgc/3508.

I guess you meant: https://discuss.px4.io/t/reboot-pixhawk-from-qgc/3508.

love-mj, 2019-07-31:
yes

julianoes, 2019-07-31:
@love-mj when RC doesn't work, could you run the command in QGC mavlink shell?

https://docs.qgroundcontrol.com/en/analyze_view/mavlink_console.html

```
px4io status
```

love-mj, 2019-07-31:
yes

julianoes, 2019-07-31:
Can you copy the status here?

love-mj, 2019-07-31:
I can't. Because I'm not familiar with  it.

julianoes, 2019-07-31:
Screenshot is fine too.

love-mj, 2019-07-31:
Okay, I'll give you a screenshot as soon as possible.

davids5, 2019-07-31:
@davids5 - test on CUBE.

julianoes, 2019-08-05:
@love-mj any chance that you could get us a screenshot? Thx.

love-mj, 2019-08-05:
I'm sorry to have waited so long, because I'm doing airsim, so I'll record a video about airsim using firmware for you.

love-mj, 2019-08-06:
I retested several times and found that I couldn't reproduce the bug. Everything was working well. The situation that I sent a reboot command after normal flight in GPS mode should have nothing to do with firmware reboot. I should restart first and there won't be any bugs.Thank you for your attention to my news.

julianoes, 2019-08-06:
Ok odd. Please just re-open the issue if it happens again.

closed 2019-08-06 by julianoes

ArkadiuszNiemiec, 2020-10-21:
@julianoes it happened few times on `1.9.2`. Is it fixed in newer versions?
We was able to trigger it by rebooting the PX4 via MAVLink command. The Radio tab in QGroundControl shows that the RCIN is working but we cannot change a state, enable kill switch etc.

This is the output you asked:
[image]