PX4/PX4-Autopilot #12406: Fixed wing cutting corners in MISSION

EnsignCrasher, 2019-07-03:
Hello again,

as I'm now able to fly missions (:tada:), I observed that the autopilot is really cutting corners because it  is heading to the next waypoint way too early probably (~60m too early to be precise).
Is that the expected behaviour? Or is there any way to shorten that distance or even configure that it shouldn't turn until hitting the waypoint?
I have to admit that I haven't touched FW_L1_PERIOD yet because I would expect that the wing turns earliest, when it is FW_L1_PERIOD away from the waypoint. That would be 20m for me.

I even have a log file here :relaxed: 
https://logs.px4.io/plot_app?log=48aa7c67-1145-4769-8e85-f699f03f5d13

bresch, 2019-07-03:
Hi @EnsignCrasher ,
Congratulations for your fist mission! :) The switch distance is defined by the L1 distance. L1 distance is defined using L1 period, L1 damping and current ground speed with the following formula:
` L1_dist = 1/pi x L1_damping x L1_period`
With default settings (period = 20, damping = 0.75, speed = 15m\s) it makes L1 distance of ~72m.

(Please use the forum for next questions like that : http://discuss.px4.io )

closed 2019-07-03 by bresch

EnsignCrasher, 2019-07-03:
ah, thank you for your great explanation! :+1: 
Alright, I'll take a look there ;)
keep up the good work!