PX4/PX4-Autopilot #12409: Sensors startup reorder and cleanups

BazookaJoe1900, 2019-07-03:
The current order of sensors start up:

- calling for common rc.sensors from main rcS
	- running some sensors that seems to be "common"
		- ms5611 (on external bus)
		- ms5611 (on internal bus)
		- Blacksheep telemetry
	- running optional sensors that are enabled by parameters like airspeed, batt_smbus, ll40ls and so on
	- Wait 20 ms for sensors (because we need to wait for the HRT and work queue callbacks to fire)
	- starting the sensors module
- calling rc.board_sensors from main rcS
	- starting per board sensors list
	
I think that there are few errors on this method
1) internal ms5611 should not be so "common" and might move to rc.board_sensors
2) external ms5611, should that also be 'common'? does all configuration uses that?
3) Blacksheep might also be under optional sensors section, (under XX_EN parameter?)
4) running the delay and the sensors module should be done after all sensors has started, eg after rc.board_sensors

Do you agree?

dagar, 2019-07-03:
Hi @BazookaJoe1900, this is a good idea and probably something we could continue to cleanup incrementally. The larger reason I haven't done more already is that we really want to get to a "board manifest" that lays these things out statically.

Something like this per board in yaml (TBD). The required sensors per board are layed out and this would both include the driver in a particular build, handle init, and perhaps most importantly report as a critical error if a required sensor is missing.

[image]

BazookaJoe1900, 2019-07-03:
@dagar  Do you want mw to do just the cleanup I suggested above?

BazookaJoe1900, 2019-10-07:
closing the issue.
Some other "side issues" that related to clean start up opened and each has his relevant status

closed 2019-10-07 by BazookaJoe1900