PX4/PX4-Autopilot #12412: thermal calibration issues

jbourne15, 2019-07-04:
I am trying to perform a [offboard thermal calibration on my pixhawk 3] with v1.9.0 firmware (https://docs.px4.io/en/advanced_config/sensor_thermal_calibration.html).  And I am getting step like response for my bias compensation (see attached figure).  Notice the z acceleration bias is very large.  Looking at the [flight review](https://review.px4.io/plot_app?log=9ef38f6d-629e-43c8-84f3-5af83118771b) it seems that the raw data is fine and has temperature drift, but the estimate rpy angle and rate go crazy.  

Why do I get step like responses for my bias curves? Is this a hardware or software problem?
 Has anybody got this issues before and if so how can I resolve it? This problem exist and my 3 new pixhawk 3 systems.

[log001.ulg.pdf](https://github.com/PX4/Firmware/files/3357190/log001.ulg.pdf)

thanks

dagar, 2019-07-04:
Interesting that it's only on the 2nd sensor (both gyro & accel). Is this reproducible?

[image]

Also note that your baro cal looks completely bogus.

jbourne15, 2019-07-04:
Yes it is repeatable. Probably tried it 4-5 times now.

jbourne15, 2019-07-04:
Does the pixhawk accelerometer and gyro need to be previously calibrated before I thermally calibrate? I am wondering if CAL_ACC1_ID needs to be set up first.

jbourne15, 2019-07-05:
I forgot to mention that I am performing the offboard calibration procedure.

jbourne15, 2019-07-09:
After talking with Drotek, it seems they have some issues with the IMU and/or IMU temperature sensor on the pixhawk 3 so this may be just a hardware issue.

closed 2019-07-09 by jbourne15

julianoes, 2019-07-09:
@jbourne15 That's strange but thanks for the note.