PX4/PX4-Autopilot #12413: VTOL: Add support for loiter before backtransition

RomanBapst, 2019-07-04:
From discussions with @eyeam3 we concluded that it would be a good idea to enhance the VTOL mission logic for the back-transition.
The idea is for the user to define a loiter waypoint at which the vehicle would circle down to the required back-transition altitude. Furthermore, the vehicle would exit the circle at a predefined angle, e.g. the tangent to the next waypoint (landing or home).
Doing so the user always knows what to expect from the back-transition and the predicted path could even be shown in QGC.
This would also solve a problem we currently have when an RTL is triggered in fixed wing mode at high altitude close to the home position. In this case the navigator ignores the altitude and transitions to hover mode immediately without first reducing altitude in fixed wing mode.
Additionally this feature would have the benefit that the system can learn the wind while it's circling which is good to know for the transition and landing phase.

Summary of benefits:
- back-transition would become very deterministic
- back-transition always starts in straight an level flight
- no more risk of transitioning into arbitrary directions
- no more risk of having to descend from high altitudes while hovering
- allows to get an accurate wind estimate prior to transitioning to hover

The following figure gives a good idea how a VTOL could attempt to approach for a back-transition. I manually planned this mission in SITL by adding a loiter waypoint before the transition and specifying "leave on tangent to next waypoint" in QGC. 

[image]

sanderux, 2019-07-06:
> In this case the navigator ignores the altitude and transitions to hover mode immediately without first reducing altitude in fixed wing mode.

I thought i recently changed that behavior and it would indeed circle down before transitioning

sanderux, 2019-07-06:
Although, in general, i think this could be a good idea i should note a few things:
1. when circling down you increase the speed which is not always optimal for back transitions
2. the circling behavior might not be clear to everyone, it could become dangerous when the vehicle flies unexpected paths at low altitude
3. how would this look if no decent (or hardly) would be required?

RomanBapst, 2019-07-06:
@sanderux Thanks for you inputs:

> when circling down you increase the speed which is not always optimal for back transitions

Once you reach the desired transition altitude you can always reduce speed, in fact this was actually the plan. The vehicle should transition as soon as it's tangent to the next waypoint and has reduced it's speed. In the worst case this would mean that it will fly another circle.

> the circling behavior might not be clear to everyone, it could become dangerous when the vehicle flies unexpected paths at low altitude

The behavior will be indicated clearly in QGC. In fact the user needs to place the circle at a suitable location and thus back-transition behavior will become very deterministic and should be displayed properly in QGC. Arbitrary excursions into arbitrary directions like we have now should not happen anymore.

> how would this look if no decent (or hardly) would be required
The vehicle would still fly towards the circle and IMO complete one turn. It gives consistent behavior and also helps to estimate the wind very precisely at the transition altitude.

taileron, 2019-07-06:
I´m currently doing experiments with momentary gliding periods in between fw and mc mode thats the most efficient way to descend consuming no battery at all

sfuhrer, 2019-08-14:
As I mentioned in https://github.com/PX4/Firmware/pull/12464, the fixed-wing landing pattern can also be used for VTOL landings and RTLs. 

[image]

As soon as we have test flown this a bit we should think about how to improve the UX for VTOL in QGC, as currently the pattern is quite hard to find, and also either rename it to FW/VTOL landing pattern or make a new dedicated VTOL landing pattern.

We could even think about that a landing pattern appears per default when "VTOL transition and land" is selected.

RomanBapst, 2019-11-21:
This is possible as of right now, closing.

closed 2019-11-21 by RomanBapst