PX4/PX4-Autopilot #12422: Unable to arm my drone

raph12345, 2019-07-06:
Hello everyone
I recently mounted a drone based pixhawk 4 (holybro) but today my drone refuses to arm and I really do not understand why.
my drone was flying very well in 1.9.0 and it seems to me that the problem occurred following the update in 1.9.1 because I have not made a parameter change in-temp.
I do not know what it can come from yet I searched the forum if solutions had been found on this kind of problem.

my gps seems to work:

[image]

below the result of a ls /dev and a ms5611 start

[image]

quoted qgroundcontrol software does not show me error

[image]

[image]

in the logs I see 2 error
[E] at :0 - “serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.”
[E] at C:\projects\qgroundcontrol\src\PositionManager\PositionManager.cpp:139 - “QGCPositionManager error 1”

but I’m not sure my problem comes from there … because I’m not sure I understand the error messages either

looking for this or this QT_NMEA_SERIAL_PORT parameter could be changed i have to try modified the serial port baudrate

[image]

and when restarting qgroundcontrol displays the error below.

[image]

here are the logs. (rename log.pdf in log.txt to see it)
[log.pdf](https://github.com/PX4/Firmware/files/3364511/log.pdf)

If anyone could take a look at it and tell me why I can not arm my drone it would be nice!

thank you

julianoes, 2019-07-08:
Is there no message when it refuses to arm?

raph12345, 2019-07-09:
Hi 
the only messages I have are those in the log file. 
I activated the logs to be able to retrieve them on the sd card  because the drone is not connected in usb at this time. 
the LED of my gps antenna is blue and when I try to arm via the switch the led flashes quickly in white and then turns blue again ... normally when the drone is armed the led is green. 
is there a way to increase the log level on the sd? 
I admit to being stuck there Thank you for your help :)

hamishwillee, 2019-07-09:
Blue means armed with no GPS lock: http://docs.px4.io/en/getting_started/led_meanings.html

I presume you have tried this outside where you know you have a great GPS signal. 

You're best off connecting to QGC via USB or a telemetry module. I don't know much about the logs.

julianoes, 2019-07-09:
Note that you can't arm with USB attached unless you disable the circuit breaker:
https://dev.px4.io/en/advanced/parameter_reference.html#CBRK_USB_CHK

julianoes, 2019-10-08:
@raph12345 is this resolved?

raph12345, 2019-10-08:
Yes you can close.

My pixhawk was Dead, after exchange its ok.

closed 2019-10-08 by julianoes