PX4/PX4-Autopilot #12423: prefligt fail baro sensor 0 missing

raph12345, 2019-07-06:
Hello,

recently I have this error message in QGC that appears when I try to arm my drone while it is connected with a USB cable to my PC.

what is this error message?

thank you

bresch, 2019-07-06:
Hello @raph12345 which autopilot hardware do you have? Some boards don't have a barometer. Can you open QGC, go to the MAVLink console and enter `sensors status'? You should directly see if there is a barometer connected and working or not.

raph12345, 2019-07-07:
hi bresh,

I use a pixhawk 4 (holybro) and the latest version of firmware 1.9.1.

below the result of the command sensor status:
[image]

I turned off the barometer and restarted and I tried to arm the drone but without result ...
[image]

in QGC where can I find the preflight cheklistlist?

hamishwillee, 2019-07-07:
> in QGC where can I find the preflight cheklistlist?

https://docs.qgroundcontrol.com/en/SettingsView/General.html#miscellaneous

"Use preflight checklist" checkbox

LorenzMeier, 2019-07-09:
Your hardware is faulty. It should have a baro, but it isn't detected. Ask the vendor for a replacement.

closed 2019-07-09 by LorenzMeier