PX4/PX4-Autopilot #12432: Parameters being used after logging/arming

garfieldG, 2019-07-08:
The following parameters are being used only after arming :
COM_ARM_EKF_AB 
COM_ARM_EKF_GB
COM_ARM_EKF_HGT
COM_ARM_EKF_POS 
COM_ARM_EKF_VEL
COM_ARM_EKF_YAW

same goes to SDLOG_UUID param, being used only after log has started.

May cause issues if you try to go over your parameters and then not even being aware there are more parameters not shown just because you didn't arm/started new log.

julianoes, 2019-07-08:
FYI @dagar

dagar, 2019-07-08:
The COM_ARM_EKF* parameters are read slightly later in commander preflight check, but still only before arming. `SDLOG_UUID` is found and used at log creation.

@garfieldG have you verified this in current master or v1.9.0? As of https://github.com/PX4/Firmware/commit/f8519781011a6d0ca3114a8854b2b8f06832a283#diff-06bd9bf93c4b14e72b5518a112c266b7 I think it should be resolved for the commander parameters.

garfieldG, 2019-07-16:
@dagar just verified it with v1.9.2 and you're right regarding COM_ARM_EKF* params, they do appear.

julianoes, 2019-07-18:
@garfieldG so this is resolved?

garfieldG, 2019-07-18:
@julianoes as to COM_ARM_EKF* params params yes.
 SDLOG_UUID founded only in log creation.

closed 2022-08-15 by bresch