PX4/PX4-Autopilot #12438: PX4 "Reject Auto hold" and "Reject Auto Mission"

FlyingToTheSpace, 2019-07-09:
Hello,
I'm using pixhawk4 mini with GPS.

I succeeded flying with manual and altitude flight mode.
However, with some reasons, I failed to change to "Hold mode" or "Mission mode" with error message "Reject Auto Hold" or "Reject Auto Mission"

I have 
GPS count: 18
GPS Lock: 3D DGPS Lock
HDOP: 0.7
VDOP: 1.3

Please help me out.
Log file attached.

[log_140_2019-7-8-15-01-44.zip](https://github.com/PX4/Firmware/files/3370942/log_140_2019-7-8-15-01-44.zip)

mcsauder, 2019-07-09:
@FlyingToTheSpace , could you upload your files to [PX4 Flight Review](https://review.px4.io/) and post the link?  This will make review very easy and might also gain you insight into what is going on!

FlyingToTheSpace, 2019-07-09:
https://logs.px4.io/plot_app?plots=pid_analysis&log=e34d00f5-147c-4891-b947-8ebba8172b2e

Thank you

mcsauder, 2019-07-09:
Hi @FlyingToTheSpace , your log analysis is a good start, but it seems to be lacking in some details.  Could you try to repeat a flight and wait to hear the tone alarm enunciator coincide with a green flashing light on your GPS unit to demonstrate that you have GPS lock, then try to switch to altitude hold mode or position mode and post a new log?  This would be very helpful in analyzing.

julianoes, 2019-07-09:
Presumably that's the same as #12168.

FlyingToTheSpace, 2019-07-10:
@mcsauder Hello, This is new log as you requested. Flight maintained 10sec for manual, 10sec for altitude, then 10sec for hold mode which was rejected. Thank you for you help.

https://review.px4.io/plot_app?log=9e026cd1-5568-40f8-9e16-89022971ce9c

mcsauder, 2019-07-10:
Hi @FlyingToTheSpace , looking at your logs it looks like @julianoes' advice was spot on. Referencing #12204 it appears that a valid home position was not set:
[image]

Here is what those same values look like on a successful mission:
[image]

FlyingToTheSpace, 2019-07-10:
How can I set valid home position?

I'm using QGroundControl and I cannot find how to set home position from user guide.

This is the picture of my ground station.
THANK YOU

[image]

mcsauder, 2019-07-10:
Hi @FlyingToTheSpace, you need to allow the GPS to get a better position lock before arming your drone. The step in the plot above from your log shows that your system eventually acquired enough position certainty, but not until long after you had armed the system. You might have to try repositioning your GPS antenna differently on your airframe or make other adjustments to your setup to help get a better gps position lock more quickly.

julianoes, 2019-07-10:
Ok so it looks the same as #12168, closing as duplicate. Thanks for reporting @FlyingToTheSpace.

closed 2019-07-10 by julianoes