PX4/PX4-Autopilot #12439: Auto-estimate hover throttle

LorenzMeier, 2019-07-09:
Setting the correct hover throttle seems to be a consistent issue for users and is a source of a common frustration. We need to implement an estimator that takes current vertical velocity and estimates hover throttle for a given system. Once it is estimated, the parameter should be set so that on next takeoff it is already correct.

This is also relevant for commercial models where people add/remove large cameras and will certainly forget to adjust the vehicle settings.

dagar, 2019-07-10:
What about the voltage drop over a flight?

bresch, 2019-07-11:
@cmic0 Can you give links to you dev branch, prototype script and ref papers of your hover throttle estimator please?

bresch, 2019-07-11:
@dagar Voltage drop compensation only works if you have it calibrated (feedforward compensation). Here we want an estimator that basically learns the hover thrust such that the user doesn't even need to specify the hover thrust via a parameter and that self-adjusts during flight.

dagar, 2019-07-11:
> @dagar Voltage drop compensation only works if you have it calibrated (feedforward compensation). Here we want an estimator that basically learns the hover thrust such that the user doesn't even need to specify the hover thrust via a parameter and that self-adjusts during flight.

I'm talking about the hover thrust changing over the course of a flight as a result of the battery discharge (on typical hobby ESCs). The estimated (and committed) hover thrust by the end of a flight will then be off on the next flight with a fresh battery. Yes it will be adjusted again, but it might not be correct at takeoff. Mainly just curious if that's a consideration.

[image]
https://logs.px4.io/plot_app?log=b33c30d5-3816-4cfd-9ca4-2296948ca7bb

[image]

Antiheavy, 2019-07-11:
Changes in density altitude (different days, different weather, different locations) could play a significant roll here as well.

jkflying, 2019-07-16:
This is also a consideration for any cargo drone with a payload.

julianoes, 2019-07-24:
Notes from the devcall (correct me if I get it wrong):
- This should probably not be implemented in the controller but in some estimator, e.g. land_detector.
- This is work in progress and will need quite some testing.
- This hopefully makes it into v1.10.

bresch, 2019-10-23:
I'll start to work again on that in about 1-2 Current WIP here: https://github.com/Auterion/Flight_Control_Prototyping_Scripts/pull/3

closed 2020-02-07 by bresch