PX4/PX4-Autopilot #12444: PX4 SITL Gazebo ROS multi-vehicle-simulation mission

arnozeelmaekers, 2019-07-09:
Hey all

So I have being working on running a project in which we will execute missions on multiple drones.
- Firmware v1.9.1
- ROS Melodic
- Ubuntu 18.04
- To launch the simulation I am using https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

The issue I am having at the moment is that when I run a mission on uav2 it runs perfectly however if I run a mission on uav0 or uav1 it will not work note that the only difference is that the ROS topics will change from for example *uav2/mavros/mission/push* to *uav1/mavros/mission/push* or *uav0/mavros/mission/push*

- Log when I launch multi_uav_mavros_sitl.launch: https://pastebin.com/WYp8wpBu
- Log when I run mission on uav2: https://pastebin.com/SKFumeDe
- Log when I run mission on uav1: https://pastebin.com/N7dLNjXe
- Log when I run mission on uav0: https://pastebin.com/XWTmH9qU
- Code I use to run the mission: https://pastebin.com/RBDg0Lpb

Any advice on what could be wrong?

lamping7, 2019-07-09:
It sounds like this is your issue: https://github.com/PX4/Firmware/issues/9046

arnozeelmaekers, 2019-07-10:
Thanks for the quick response.

It sounds related but in my case uav0 and uav1 won't even takeoff. Does this have to do with issue #9046 or is it different. If it has to do with issue #9046 is there a fix for it?

lamping7, 2019-07-10:
#12415 then possibly

lamping7, 2019-07-10:
https://github.com/PX4/Firmware/pull/12415 was just merged in. Try master and report back.

arnozeelmaekers, 2019-07-11:
Oke so I've tested a few scenarios:
- uav0 running a mission when no other drones are flying: works
- uav1 running a mission when no other drones are flying: does not work, does not take-off
- uav2 running a mission when no other drones are flying: works

- uav0 running a mission when there are other drones flying: does not work, does not take-off
- uav1 running a mission when there are other drones flying: does not work, does not take-off
- uav2 running a mission when there are other drones flying: does work, uav0 has to take-off first

lamping7, 2019-07-11:
What does "does not takeoff" mean? What's the error produced?

arnozeelmaekers, 2019-07-11:
Trying to send uav1 on a mission results in :
Note the mission that get's send works on uav0 and uav2
```
[ERROR] [1562834769.354347826, 537.920000000]: FCU: Data link lost
[ INFO] [1562844011.154229672, 6800.436000000]: WP: mission sended
[ WARN] [1562844011.185395820, 6800.452000000]: CMD: Unexpected command 176, result 0
WARN  [navigator] No valid mission available, refusing takeoff.
WARN  [commander] Failsafe enabled: no datalink
INFO  [navigator] RTL HOME activated
```
Trying to let uav0 take-off when uav2 is already flying returns:
```
[ WARN] [1562844622.611741309, 79.916000000]: CMD: Unexpected command 176, result 0
[ INFO] [1562844622.615952153, 79.920000000]: FCU: Executing Mission
[ INFO] [1562844622.667755119, 79.972000000]: FCU: Takeoff to 5.0 meters above home.
[ INFO] [1562844622.716205052, 80.020000000]: FCU: Takeoff to 5.0 meters above home.
[ INFO] [1562844623.622761478, 80.924000000]: FCU: ARMED by Arm/Disarm component command
INFO  [commander] Takeoff detected
[ INFO] [1562844624.856019665, 82.156000000]: FCU: Takeoff detected
```

But uav0 never takes-off a bit later in the console log it returns this:
```
INFO  [commander] Landing detected
[ INFO] [1562844675.288013922, 132.552000000]: FCU: Landing detected
[ INFO] [1562844675.351766315, 132.616000000]: FCU: DISARMED by Auto disarm on land
[ INFO] [1562844675.388201654, 132.652000000]: FCU: Executing Mission
[ INFO] [1562844675.435533500, 132.700000000]: FCU: Takeoff to 5.0 meters above home.
INFO  [logger] closed logfile, bytes written: 7377939
```
*Full log of uav0 trying to take of after uav2: https://pastebin.com/XwcZ6jAP*

lamping7, 2019-07-11:
You're not giving me enough info here. I just ran master. They all go, however they all perform the same mission because of #9046.
[image]

lamping7, 2019-07-11:
It wouldn't hurt to read https://github.com/PX4/Firmware/pull/12306#issuecomment-503809590

arnozeelmaekers, 2019-07-12:
Oke so I have found the issue. It was giving errors that something was not calibrated so I had to remove parameters in the eeprom folder. 
```rm ~/.ros/eeprom/parameters```

This just leaves #9046 as an issue.

closed 2019-07-15 by lamping7