PX4/PX4-Autopilot #12447: question about crash logs

kyumanlee84, 2019-07-09:
Hello, 

I have a quick question about crash logs.
I did offboard control 

Thanks!

julianoes, 2019-07-15:
Can you check why this was triggered?
https://github.com/PX4/Firmware/blob/1fb80612f386d3bf9d1cc0edb855e7fdeb31dc69/src/modules/mavlink/mavlink_receiver.cpp#L1263

And please upload a logfile to logs.px4.io, otherwise it's hard to see what is going on.

julianoes, 2019-10-14:
Please re-open if you can provide the details we asked for.

closed 2019-10-14 by julianoes

greymaner, 2023-04-04:
> Can you check why this was triggered? https://github.com/PX4/Firmware/blob/1fb80612f386d3bf9d1cc0edb855e7fdeb31dc69/src/modules/mavlink/mavlink_receiver.cpp#L1263
> 
> And please upload a logfile to logs.px4.io, otherwise it's hard to see what is going on.

Hello, I faced this problem . when I add gcs_url:=tcp://<server-ip>:5760   in gazebo .
I generated more than one vehicle in gazebo, then it happend...

`<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:24540@localhost:34580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="18570"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value="tcp://101.200.195.227:5760"/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    
    <group ns="uav1">
        
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:24541@localhost:34581"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="18571"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value="tcp://101.200.195.227:5760"/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    
</launch>
`