PX4/PX4-Autopilot #12454: New driver I2C not initialised on px4fum-v3

w407022008, 2019-07-10:
Hi,

I was trying to create a new sensor driver with Pixhawk 2.1 (px4fmu-v3), which is mainly copied from sf1xx driver. I have already created the relative topic and added the name in default.cmake. Everything was prepared well, but some wrong and never varied data showed in the nsh console, when i called the command "sfm test" (sfm is the driver name.). 

However, if I used Arduino to drive the sensor, everything went well. If I drive the sensor by Pixhawk 2.1 and then unplug the cable SDA, and soon drive it by Arduino, and finally re-plug in the cable SDA, those data shown in nsh console went well again (They are connected in series and same GND). So I guessed that maybe the initialisation of I2C failed.

Does any friend know the reason and how to solve it?

In addition, What is the difference between PX4_I2C_BUS_EXPANSION and PX4_I2C_BUS_ONBOARD? cause when i set the i2c bus as PX4_I2C_BUS_EXPANSION, it shows me " i2c initialisation failed!", but PX4_I2C_BUS_EXPANSION it shows well, even the data are wrong also.

mcsauder, 2019-07-12:
Hi Max, the [mappydot driver](https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/mappydot/MappyDot.cpp), is a good example of a recently added i2c driver that might help you to reference.

Make sure you are adding your driver correctly in all relevant CMakeLists.txt files, [in the driver directory](https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/mappydot/CMakeLists.txt), and [in the higher level directories](https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/CMakeLists.txt).  If all of that is in order, you'll need to examine the specific usage of the [I2C class](https://github.com/PX4/Firmware/blob/master/src/lib/drivers/device/posix/I2C.cpp) methods you are utilizing, there are differences in method usage between PX4 and Arduino.

mhkabir, 2019-07-12:
Please use discuss.px4.io for questions.

closed 2019-07-12 by mhkabir