PX4/PX4-Autopilot #12463: ekf2 + vision error

lgh5054, 2019-07-11:
**Describe the bug**

When viewing the mavros/local_position/pose value, Even if you stay still, you lose your position.

Is this a slam problem or a connection speed problem with the pixhawk and tx2 board?

At first, it shows the correct position value, but after a while, x y is fixed at zero

There are many of these questions, but all of them have not been solved.

I installed the firmware version of px4fmu_v2 1.8.0 

And ekf2_aid_mask 8
sys_companion (921600)
ekf2_eva_Noise 0.20
ekf2_evp_Noise 0.20
ekf2_ev_delay 50.0ms
I watched the paramedics

**To Reproduce**
Steps to reproduce the behavior:
1. Slam running
2. roslaunch mavros px4.launch

**Expected behavior**
Why is the x y value initialized to zero?

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=18d1669c-2906-405f-ad07-a0a8757d27ae
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

lgh5054, 2019-07-11:
https://logs.px4.io/plot_app?log=3875fcb0-1798-4973-8213-e9a74f420a3e#Nav-Sampling-Regularity-of-Sensor-Data

booting log

julianoes, 2019-07-15:
@mhkabir :)

JenaEmz, 2019-08-20:
I just figure that out.The solve is using vision position directly,don't fuse it with ekf2. My drone fly much stable now. I don't think they fuse vision by a proper way. You just need to use vision position in your out control loop

closed 2019-08-20 by julianoes

gyronikeleg, 2020-04-23:
@JenaEmz Hello Jena, I am having a very similar issue.  I saw that you were able to resolve it, but I dont understand your resolution.  Can you expand upon that a bit?