PX4/PX4-Autopilot #12465: EKF2 fuses vision input but drifts in around a minute

echoGee, 2019-07-12:
Using an external vision sensor (HTC Vive) to provide position estimate to pixhawk EKF2. 
The EKF fuses the vision information for a few tens of seconds, but the estimate starts drifting and resets to the x=y=~0 origin. 
The logs of a test are here: https://logs.px4.io/plot_app?log=f16ec023-ecad-43cc-975e-2e7bfaf7deca . 
The specific test sequence was
Pitch up once 
Roll right thrice
Yaw right 4 times. 

A video of Rviz can also be here: https://streamable.com/m1201

julianoes, 2019-07-15:
@mhkabir @TSC21 is this something you can assess?

mhkabir, 2019-07-15:
Sounds like either incorrect coordinate frames, or incorrect timing (either EKF2_EV_DELAY setting, or just the timestamping).

echoGee, 2019-07-15:
If you check the rviz video, coordinate frames seems to be pretty much aligned correct. The times are also synced to maybe within 10ms or close in the logs.

mhkabir, 2019-07-15:
> The times are also synced to maybe within 10ms or close in the logs.

echoGee, 2019-07-15:
Plotted the peaks of pitch after i pitch up and down and compared with mocap data. 
There is a little bit of jitter in the data, hence the within 10ms comment.

closed 2022-08-15 by bresch