PX4/PX4-Autopilot #12469: SITL: Preflight Fail : Compass #0 uncalibrated

wuxianshen, 2019-07-12:
My px4 sitl with jmavsim and gazebo was all ok to use, and today I update source code to newest commit and then make again (sudo make px4_sitl_default jmavsim & sudo make px4_sitl gazebo).
After make the two sitls platforms prompts "SITL: Preflight Fail : Compass #0 uncalibrated " and cannot takeoff...
Is this problem normal with SITL? How to calibrate compass with QGC ( since I found it impossible to rotate the vehicle in SITL environment... )?  

Thanks!

bkueng, 2019-07-15:
Can you try a `make clean` first?
The SITL sensor ID's changed at some point.

wuxianshen, 2019-07-15:
@bkueng Yeah It worked! Thank you very much!

closed 2019-07-15 by wuxianshen

mortenfyhn, 2019-10-14:
For future reference: I had the same issue, and `make clean` (or `make distclean`) did not help, but deleting `~/.ros/eepros/parameters` worked as mentioned [here](https://discuss.px4.io/t/ros-and-gazebo-sitl-fail-v1-9-0-mavros-posix-sitl-launch-preflight-fail-compass-uncalibrated/11489).

wildwilly18, 2021-03-13:
So I can run a single quad but when I try to run two using the tutorial on https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html I am getting the issue described here. Any tips on how I can get this to work for me?

otmanebenhmimid, 2021-06-10:
please how i can solve this error in PX4 using ros and mavros in launch file 
i put this 
WARN  [PreFlightCheck] Preflight Fail: Accel #0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #0 uncalibrated
WARN  [PreFlightCheck] system power unavailable

avi0000, 2021-06-25:
> please how i can solve this error in PX4 using ros and mavros in launch file
> i put this when i tape commander takeoff
> WARN [PreFlightCheck] Preflight Fail: Accel #0 uncalibrated
> WARN [PreFlightCheck] Preflight Fail: Gyro #0 uncalibrated
> WARN [PreFlightCheck] system power unavailable

did u find any solution I am also getting the same issue

jmurraylouw, 2021-07-03:
Here are some solutions you could try: (https://github.com/Jaeyoung-Lim/mavros_controllers/issues/155#issuecomment-705714040)