PX4/PX4-Autopilot #12471: Low takeoff is not getting executed

julianoes, 2019-07-12:
**Describe the bug**
When the takeoff altitude is set too low (< than around 0.7 meters) using the param `MIS_TAKEOFF_ALT` a quadcopter won't takeoff but just auto-disarm again.

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL jMAVSim or Gazebo
2. `param set MIS_TAKEOFF_ALT 0.5`
3. `commander takeoff`

**Expected behavior**
It should still take off even if the altitude is quite low.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=b5802f74-4a52-42f2-8f31-0bbf01fa84e9

**Additional context**
This currently makes the MAVSDK integration test `ActionTakeoffAndKill` fail.

julianoes, 2019-07-12:
`git bisect` points to bb055fdaf3d1d846538d296fa664eecca2f68798. @bresch  :sob:

julianoes, 2020-02-12:
@MaEtUgR that's what I was referring to.

bresch, 2020-03-04:
@julianoes We changed quite a bit the waypoint tracking logic since then, could you check is this issue is still there?

julianoes, 2020-03-05:
@bresch nice! I tested 0.5m and 0.3m which worked. 0.2m worked sometimes and 0.1m stayed on the ground but that's already much better!

bresch, 2020-03-05:
@julianoes BTW, why is the Navigator not complaining?
[MIS_TAKEOFF_ALT](https://dev.px4.io/master/en/advanced/parameter_reference.html#MIS_TAKEOFF_ALT) is the lowest altitude you should be able to go during takeoff: 
https://github.com/PX4/Firmware/blob/b9fab04adbfd92783206a08364c8fea077f1ed39/src/modules/navigator/takeoff.cpp#L86-L105
Are you setting that parameter differently or are you not passing through this logic?

julianoes, 2020-03-05:
I'm using this parameter to set the takeoff altitude before a takeoff.

closed 2024-12-10 by julianoes