PX4/PX4-Autopilot #12475: Jenkins intermittant failure (Catkin build on ROS workspace)

dagar, 2019-07-14:
The "Catkin build on ROS workspace" job in the top level Jenkins pipeline has started failing intermittently (both PRs and master).

[image]

Typical failure output.

[image]

@TSC21 could you take a look?

lamping7, 2019-07-15:
I pulled the docker image, ran the same steps as Jenkins and ran the test numerous times without failure. Let's try to catch something in a log.

lamping7, 2019-07-15:
It seems that px4 is shutting down when it shouldn't. #12483

lamping7, 2019-07-16:
This looks good now.

closed 2019-07-16 by lamping7