PX4/PX4-Autopilot #12476: EKF2 offboard position control doesn't work with px4 firmware 1.9.2

EvanAndE, 2019-07-14:
**Expected behavior**
The goal is to do offboard position control, simply to make the quad take off from the groud to a set point (0,0,0.5) and hover. But it failed.
**Log Files and Screenshots**
https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html
[image]
The .ulg file is shown here.
https://logs.px4.io/plot_app?log=e9b8b3cb-be93-4ef5-973c-5f1e553b19a3

1. I updated the latest px4 firmware 1.9.2 through QGC. The pixhawk is fmu-v2, I use a companion compute board insatlled ros kinetic and mavros. And I'm trying to do offboard control indoor with vicon device.
2. As the  screenshot shows, I publish the vicon data to the mavros/vision_pose/pose topic, and the left window shows the data with commander line: rostopic echo mavros/vision_pose/pose, the middle window shows the data with commander line: rostopic echo mavros/local_position/pose, and the right window is the QGC analyzer, and the three sets of position data is consistent. So I think I set the EKF2 parameters in QGC right based on this 
3. After I keep publishing data through the topic mavros/vision_pose/pose and run the script which continues to publish the waypoint to the topic /mavros/setpoint_position/local, I can change the drone to the offboard mode by using QGC,  when the drone is in offboard mode, it just takes off and seems not be able to reach the fixed point and hover, which should be the position (0,0,0.5) as I set in my code,  it just drifted away in one direction, and seems to rise up constantly. I also tried the code here, https://github.com/mzahana/px4_indoor_navigation/blob/master/scripts/px4_offbaord_test.py
and it has the same results and can not hold position. I tuned the yaw rate controller PID, roll and pitch controller PID and they are fine. But I don't know whether this is a firmware issue or I need to tune the position controller PID.

julianoes, 2019-07-18:
@EvanAndE thanks for the detailed issue. Let me ping @bresch and @MaEtUgR.

julianoes, 2019-10-17:
Also mentioning @Jaeyoung-Lim.

Jaeyoung-Lim, 2019-10-20:
@EvanAndE I believe this is not a problem related to offboard control, but more likely that the external vision input you are providing leads to a wrong estimated position of the vehicle. Have you checked all the frames and `EKF_AID_MASK` set correctly as done in this [Documentation](https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html)?

bresch, 2022-08-15:
No answers, closing

closed 2022-08-15 by bresch