PX4/PX4-Autopilot #12484: PX4 V1.9.0 and Later: Attitude Control and Takeoff Issues

ar1040, 2019-07-15:
@mcsauder as you requested. I have a fleet of quads that have been flying extremely well for months now. I am using a Pixhawk Cube with a Here+ GPS. I updated one of these units to v1.9.0, and this is where I found this problem. I have also flashed v1.9.1 and v1.9.2 all with the same problems:

Takeoff: when the vehicle lifts off it seems to do so very weakly, as opposed to taking off quickly. About 2 out of 3 times, the vehicle looses heading (leaving the ground) and corrects itself during ascent, or as it reaches the takeoff height.  The issue for me, is my quads often carry heavy flat payloads in high wind situations and I am afraid this may cause a crash.

Attitude control: The vehicle is controlled via companion computer relaying to a ground station allowing easy keyboard controls. We had our craft slowed down to restrict jerky movement and to restrict yawing speeds (as I said heavy flat payloads). When I updated to v1.9.0 and up, the yawing became extremely fast (almost catastrophic). Also jerking the vehicle side to side was also very stable prior to updating, but with the newest update version, the drone almost flips itself when jerking side to side (without a payload).

I have tried this on 4 different units all with the same issue.

I would prefer to keep my vehicles up to date, but I have been sticking with v1.8.2 until the issues have been resolved.

julianoes, 2019-07-15:
@MaEtUgR @bresch is this a regression with smooth takeoff?

mcsauder, 2019-07-16:
See also #12278

ar1040, 2019-07-16:
If any other resources (ie. logs, videos, etc) are needed let me know and I will record and upload.

mcsauder, 2019-07-16:
If you can record a flight log on your system with 1.8 and another with 1.9.2 it would be helpful.  I will do the same on my system and we can look through them as an A/B comparison.

ar1040, 2019-07-17:
Of course when I reach out to the Devs my craft wants to takeoff fine. I had no yawing on takeoff on the first drone. I will try a couple more today in order to replicate the problem. However, on my last takeoff, the drone disarmed itself before/at the takeoff height. I thought I would mention this as it still has to do with takeoff. Here is the log for that crash:

https://logs.px4.io/plot_app?log=2785e143-48b4-4b78-b85c-692afe2a79a1

mcsauder, 2019-07-17:
@ar1040 , from your log above it appears that at the transition from offboard control back the autopilot thought the vehicle was landed.  I'm not familiar enough with offboard control to know what could have happened in that transition.
[image]

If you can, post a log with the attitude issue when you encounter it again and if you can get a video recording that couldn't hurt to look at either.  I cannot reproduce the issues you have described with 1.9.2 or current master, so I'm in the dark about what could be the cause.

ar1040, 2019-07-18:
@mcsauder , I did have a chance to test the attitude on that drone before it crashed. 

https://logs.px4.io/plot_app?log=63e9657f-7457-4fba-b6c8-d3ce86af8683
https://logs.px4.io/plot_app?log=326d4b5c-4e40-4e69-a5e0-cc342e1d0bd9
https://logs.px4.io/plot_app?log=9a29f67a-955c-402f-93f3-cc33f577e416

I gave the craft multiple maneuvers while testing. There are some hard side-to-side maneuvers included. During these hard side-to-side manuvers, the drone acted like I described earlier, acting as it may flip, and becoming overly responsive. Looking through the logs, I couldn't see much of a difference in the roll angle like I expected, but maybe you could see more than I can.

ar1040, 2019-07-18:
I also wanted to add, I mistakenly had a drone with the version 1.9.0, and the same thing happened, it fell from the sky during takeoff. This has not been a problem for me when using 1.8.2, and I am not sure of the differences. Here's the log to that crash to see if there is any consistency:

https://logs.px4.io/plot_app?log=0ac3974c-0501-44b9-98b4-f70a4ec885c7

tonight123, 2019-10-23:
I have the same hardware, firmware, as you have, and I have the same problem。
takeoff： mpc_tko_ramp_t =0,it's going to take off quickly.
Starting counterclockwise after take-off, probably because rate i is too big, i is currently 0.2

julianoes, 2020-01-24:
@ar1040 were you able to get to the bottom of this or fix it?

ar1040, 2020-01-24:
Unfortunately, we had to mode on. We stuck to using the latest commit of v1.8.2. This version seems to have minimal issues. I will be looking forward to testing v1.10 in the near future, and I will reopen this if the problem persists. Thank you all for your help.

closed 2020-02-12 by julianoes