PX4/PX4-Autopilot #12485: critical: string formatting bug leading to invalid ulog files

bkueng, 2019-07-15:
On master the following code:
```
	char b[50];
	size_t s = 5;
	snprintf(b, sizeof(b), "char[%zu] %s", s, "test");
	printf("b=%s\n", b);
	printf("char[%zu] %s\n", s, "test");
	snprintf(b, sizeof(b), "%zu", s);
	printf("b=%s\n", b);
	printf("%zu\n", s);
```
produces:
```
b=char[%zu] test
char[%zu] test
b=%zu
%zu
```
on NuttX. The expected output is:
```
b=char[5] test
char[5] test
b=5
5
```

As a result **all** (!) log files are invalid. The reason why it still works is that `pyulog` does not look at that field for string lengths, but FlightPlot for instance won't open the log.

`%zu` is both a valid C99 and C11 format specifier. It is used in the logger and at a few other places.

The commit that breaks it is the NuttX upgrade (dc10a685392f9c45e300fd76fb61a80e8669a8d7).

@davids5 can you have a look?

davids5, 2019-07-22:
@bkueng - The change was introduced in the new stdio/lib_libvsprintf.c. The legacy_libvsprintf is still available but uses more space. Let me see if I can upstream a fix and backport it.

davids5, 2019-07-22:
@bkueng - PR submitted upstream - I will back port if it goes in or added at as a reject if not.

davids5, 2019-10-21:
Fixed in https://github.com/PX4/NuttX/pull/56

closed 2019-10-21 by davids5