PX4/PX4-Autopilot #12486: Safety protocol after losing GPS while in offboard mode.

jbourne15, 2019-07-16:
Can someone clear this up for me?  What happens when flying in offboard mode and relying on gps and then suddenly fall below acceptable levels?  I don't think this is very clear in the [documentation](https://docs.px4.io/v1.9.0/en/config/safety.html). The Offboard Loss Failsafe section doesn't specify what happens in this case.

bozkurthan, 2019-07-17:
Hello @jbourne15 ,
Actually it's mentioned in following lines:

> If a failsafe occurs while the vehicle is responding to another failsafe (e.g. Low battery while in Return mode due to RC Loss), the specified failsafe action for the second trigger is ignored. Instead the action is determined by separate system level and vehicle specific code. This might result in the vehicle being changed to a manual mode so the user can directly manage recovery.

Furthermore, in my opinion you can try in simulation.
As far as I know, if position can't be estimated by estimator anymore relying on GPS, current logic try to estimate position by using Optical Flow,  vision position fusion etc. which are relevant `EKF2_AID_MASK` parameter configuration. It's landing its current position if cannot estimate its position in default.

> Position (GPS) Loss Failsafe
> The Position Loss Failsafe is triggered if the quality of the PX4 position estimate falls below acceptable levels (this might be caused by GPS loss) while in a mode that requires an acceptable position estimate.
> 
> The failure action is controlled by COM_POSCTL_NAVL, based on whether RC control is assumed to be available (and altitude information):
> 
> 0: Remote control available. Switch to Altitude mode if a height estimate is available, otherwise Stabilized mode.
> 1: Remote control not available. Switch to Land mode if a height estimate is available, otherwise enter flight termination.
> Fixed Wing vehicles additionally have a parameter (NAV_GPSF_LT) for defining how long they will loiter (circle) after losing position before attempting to land.

jbourne15, 2019-07-17:
Thanks @bozkurthan for the reply and the info.  Would flying in offboard mode and then losing GPS be considered as having two failsafe protocols triggered or just one?   I would like to try this in simulation, how would I shutoff the gps during flight?

bozkurthan, 2019-07-17:
You can easily use " **gpssim stop** " on px4 terminal that you run simulation.

> Would flying in offboard mode and then losing GPS be considered as having two failsafe protocols triggered or just one? 

For that case Position (GPS) Loss Failsafe will be triggered another failsafe won't be triggered until conditions are met (RC LOSS, DATALINK LOSS etc.). Offboard loss meaning you lost connection with another node that you use for attitude control of vehicle.

If a failsafe occurs while the vehicle is responding to another failsafe (e.g. Low battery while in Return mode due to RC Loss), the specified failsafe action for the second trigger is ignored. 

But you can find more discuss about this on [https://discuss.px4.io/]().

jbourne15, 2019-07-17:
@bozkurthan I am currently using ROS to interface px4 with a gazebo simulator as described [here](https://dev.px4.io/v1.8.0/en/simulation/ros_interface.html).  And when I start my simulation I can't use "gpssim stop" because it isn't a px4 shell.  Is there a way to shutoff the gps with a ROS interface gazebo simulator?

bozkurthan, 2019-07-18:
Yeap, you can use qgroundcontrol mavlink console for the same command. (https://docs.qgroundcontrol.com/en/analyze_view/mavlink_console.html)

julianoes, 2019-07-18:
@jbourne15 is this clear? And if not would you mind making a pull request to the docs to improve it? Thanks.

jbourne15, 2019-07-18:
@julianoes Yes this is clear.

@bozkurthan thanks, I will try this.

closed 2019-07-18 by jbourne15